RMerl/asuswrt-merlin.ng #178: DDNS and Web Access Link for Alexa and IFTTT Integration not working

sectoramen, 2018-07-30:
Operation Mode:Wireless router 
Firmware Version:384.6

The **here** link in: "_Please click **here** to enable DDNS and Web Access from WAN for remote control_" does not do anything on both:
https://192.168.1.1:8443/Advanced_Smart_Home_Alexa.asp
https://192.168.1.1:8443/Advanced_Smart_Home_IFTTT.asp

Tested on Chrome and Safari

RMerl, 2018-08-17:
Works for me.  It showed me a EULA, then an alert() asking me to confirm I wanted to enable it,then the page refreshed itself.

mazafra1, 2020-09-11:
Same here.

RT-AC86U
Operation Mode:Wireless router
Firmware Version: 384.18

The here link in: "Please click here to enable DDNS and Web Access from WAN for remote control" does not do anything on both:
https://192.168.1.1:8443/Advanced_Smart_Home_Alexa.asp
https://192.168.1.1:8443/Advanced_Smart_Home_IFTTT.asp

HTTP 404 Not found

Tested on Chrome and Safari

closed 2020-09-21 by RMerl