RMerl/asuswrt-merlin.ng #179: latest test version of dnsmasq

DowngradeTheKernel, 2018-07-31:
@RMerlin @themiron the latest version of dnsmasq is 2.80test3 (release 21/Jul/2018)
http://www.thekelleys.org.uk/dnsmasq/test-releases/?C=M;O=D

themiron, 2018-08-01:
@DowngradeTheKernel thanks, it has several cache improvements on top of currently used version

themiron, 2018-08-01:
updated with eb531a072ccc730dde9ea77a46887bd2db07daf6

closed 2018-08-01 by themiron

DowngradeTheKernel, 2018-08-01:
Thank you.

**dnsmasq** snapshot (just to see, if there is an important fix)
http://thekelleys.org.uk/gitweb/?p=dnsmasq.git;a=summary

DowngradeTheKernel, 2018-08-01:
Thank you.

**dnsmasq** snapshot (just to see, if there is an important fix)
http://thekelleys.org.uk/gitweb/?p=dnsmasq.git;a=summary
http://thekelleys.org.uk/gitweb/?p=dnsmasq.git;a=shortlog

DowngradeTheKernel, 2018-08-01:
Thank you. http://thekelleys.org.uk/gitweb/?p=dnsmasq.git;a=summary http://thekelleys.org.uk/gitweb/?p=dnsmasq.git;a=shortlog
--

DowngradeTheKernel, 2018-08-01:
Thank you. 

http://thekelleys.org.uk/gitweb/?p=dnsmasq.git;a=summary 
http://thekelleys.org.uk/gitweb/?p=dnsmasq.git;a=shortlog