RMerl/asuswrt-merlin.ng #180: DHCP Server DNS Fails

sectoramen, 2018-08-01:
RT-AC5300
Operation Mode: Wireless router 
Firmware Version: 384.6

I have PiHole running on a RaspberryPi which is connected via wire to one of the router's LAN ports. I want to use it as my DNS server for internal hosts. When configuring the router's LAN->DHCP SERVER->DNS Server 1 to point to the IP of the RaspberryPi, dnsmasq fails to start. The following setting are configured:

DNS Server 1 | 192.168.1.100 (RaspberryPI IP)
DNS Server 2 | EMPTY
Advertise router's IP in addition to user-specified DNS | No
Forward local domain queries to upstream DNS | No
Enable DNSSEC support | No
Enable DNS Rebind protection | No
WINS Server | EMPTY

Log

Jul 31 22:09:22 kernel: br0: received packet on wl0.1 with own address as source address
Jul 31 22:09:23 rc_service: watchdog 371:notify_rc start_dnsmasq
Jul 31 22:09:24 dnsmasq[32267]: bad IPv6 address at line 25 of /etc/dnsmasq.conf
Jul 31 22:09:24 dnsmasq[32267]: FAILED to start up
Jul 31 22:09:53 rc_service: watchdog 371:notify_rc start_dnsmasq
Jul 31 22:09:54 dnsmasq[32297]: bad IPv6 address at line 25 of /etc/dnsmasq.conf
Jul 31 22:09:54 dnsmasq[32297]: FAILED to start up
Jul 31 22:10:23 rc_service: watchdog 371:notify_rc start_dnsmasq
Jul 31 22:10:24 dnsmasq[32326]: bad IPv6 address at line 25 of /etc/dnsmasq.conf
Jul 31 22:10:24 dnsmasq[32326]: FAILED to start up
Jul 31 22:10:53 rc_service: watchdog 371:notify_rc start_dnsmasq
Jul 31 22:10:54 dnsmasq[32352]: bad IPv6 address at line 25 of /etc/dnsmasq.conf
Jul 31 22:10:54 dnsmasq[32352]: FAILED to start up
Jul 31 22:11:24 rc_service: watchdog 371:notify_rc start_dnsmasq
Jul 31 22:11:25 dnsmasq[32381]: bad IPv6 address at line 25 of /etc/dnsmasq.conf
Jul 31 22:11:25 dnsmasq[32381]: FAILED to start up
Jul 31 22:11:54 rc_service: watchdog 371:notify_rc start_dnsmasq
Jul 31 22:11:55 dnsmasq[32404]: bad IPv6 address at line 25 of /etc/dnsmasq.conf
Jul 31 22:11:55 dnsmasq[32404]: FAILED to start up

RMerl, 2018-08-01:
Something is wrong with your IPv6 configuration.

Please use the support forums at SNBForums.

closed 2018-08-01 by RMerl

sectoramen, 2018-08-01:
Eric,

IPv6 information is auto-configured and received by the ISP with no manual
entry on my part, see below. I removed the actual IPv6 addresses. IPv6
works fine for me. Are you saying the ISP is sending the wrong information,
if so, why would dnsmasq not work only when I change the DHCP DNS settings
to point to an IPv4 internal address? I'll check with SNBForums.

[image: Screen Shot 2018-08-01 at 12.23.54 PM.png]

RMerl, 2018-08-01:
I don't know if it's the ISP sending the wrong info (which would surprise me) or an invalid setting stored in nvram.  You'd have to post your dnsmasq.conf content to determine what's on that line 25.  That's why I prefer such troubleshooting to be done on the forums, where more eyes can peek at it - my ISP doesn't support IPv6, so my ability to test anything related to IPv6 is very limited.

themiron, 2018-08-01:
@sectoramen please login via telnet/ssh and show "cat /etc/dnsmasq.conf" output

sectoramen, 2018-08-01:
This is the one after I change the DHCP DNS to point to my RaspberryPi
running PiHole.

user@RT-AC5300-AB40:/tmp/home/root# more /etc/dnsmasq.conf

pid-file=/var/run/dnsmasq.pid

user=nobody

bind-dynamic

interface=br0

interface=pptp*

no-dhcp-interface=pptp*

no-resolv

servers-file=/tmp/resolv.dnsmasq

no-poll

no-negcache

cache-size=1500

min-port=4096

bogus-priv

domain-needed

dhcp-range=lan,192.168.1.2,192.168.1.254,255.255.255.0,86400s

dhcp-option=lan,3,192.168.1.1

dhcp-option=lan,6,192.168.1.92

dhcp-option=lan,252,"\n"

ra-param=br0,10,600

enable-ra

quiet-ra

dhcp-range=lan,::,constructor:br0,ra-stateless,64,600

dhcp-option=dnsf5,option6:23,[2a02:6b8::feed:bad],[2a02:6b8:0:1::feed:bad]

dhcp-option=dnsf6,option6:23,[2a02:6b8::feed:a11],[2a02:6b8:0:1::feed:a11]

dhcp-option=dnsf11,option6:23,[192.168.1.92]

dhcp-option=lan,option6:23,[::]

dhcp-authoritative

interface=tun21

interface=tun22

addn-hosts=/etc/hosts.dnsmasq

dhcp-host=[REMOVED]

dhcp-host=[REMOVED]

dhcp-host=[REMOVED]

dhcp-host=[REMOVED]

user@RT-AC5300-AB40:/tmp/home/root#

Here is the resulting log entry:

Aug  1 17:35:12 dnsmasq[8904]: bad IPv6 address at line 25 of
/etc/dnsmasq.conf
Aug  1 17:35:12 dnsmasq[8904]: FAILED to start up

This is the working one after I removed the DHCP DNS1

user@RT-AC5300-AB40:/tmp/etc# cat /etc/dnsmasq.conf

pid-file=/var/run/dnsmasq.pid

user=nobody

bind-dynamic

interface=br0

interface=pptp*

no-dhcp-interface=pptp*

no-resolv

servers-file=/tmp/resolv.dnsmasq

no-poll

no-negcache

cache-size=1500

min-port=4096

bogus-priv

domain-needed

dhcp-range=lan,192.168.1.2,192.168.1.254,255.255.255.0,86400s

dhcp-option=lan,3,192.168.1.1

dhcp-option=lan,252,"\n"

ra-param=br0,10,600

enable-ra

quiet-ra

dhcp-range=lan,::,constructor:br0,ra-stateless,64,600

dhcp-option=dnsf5,option6:23,[2a02:6b8::feed:bad],[2a02:6b8:0:1::feed:bad]

dhcp-option=dnsf6,option6:23,[2a02:6b8::feed:a11],[2a02:6b8:0:1::feed:a11]

dhcp-option=lan,option6:23,[::]

dhcp-authoritative

interface=tun21

interface=tun22

addn-hosts=/etc/hosts.dnsmasq

dhcp-host=[REMOVED]

dhcp-host=[REMOVED]

dhcp-host=[REMOVED]

dhcp-host=[REMOVED]

user@RT-AC5300-AB40:/tmp/home/root#

and this is the log entry when it is working:

Aug  1 17:38:04 dnsmasq[12372]: read /etc/hosts - 10 addresses
Aug  1 17:38:04 dnsmasq[12372]: read /etc/hosts.dnsmasq - 4 addresses
Aug  1 17:38:04 dnsmasq[12372]: using nameserver 75.75.76.76#53 for domain
hsd1.tn.comcast.net
Aug  1 17:38:04 dnsmasq[12372]: using nameserver 75.75.75.75#53 for domain
hsd1.tn.comcast.net
Aug  1 17:38:04 dnsmasq[12372]: using nameserver 75.75.76.76#53
Aug  1 17:38:04 dnsmasq[12372]: using nameserver 75.75.75.75#53