RMerl/asuswrt-merlin.ng #182: [Suggestion] add kernel module exFAT and

tsingui, 2018-08-11:
asus router does‘t support exFAT filesystem.
 USB flash fat32 filesystem does not support 4GB files
aclean sources: https://github.com/dorimanx/exfat-nofuse

crkinard, 2018-08-11:
I thought this required licensing fees paid to Microsoft.

To add exFAT support to Synology decides it costs $4 per device.

RMerl, 2018-08-11:
Not gonna happen, for multiple reasons.

1) Asuswrt kernel is too old
2) exFat is protected by patents owned by Microsoft
3) Would require also implementing support for it at the firmware level

closed 2018-08-11 by RMerl

l-g, 2019-01-01:
There is a free (OSS) exFat implementation here https://github.com/relan/exfat

Anything specific that prevents it from being used?

RMerl, 2019-01-02:
Already explained in https://github.com/RMerl/asuswrt-merlin.ng/issues/182#issuecomment-412250918 .

l-g, 2019-01-02:
I am not sure I follow you now. The linked port is a reversed engineered OSS-solution so no patents issues. It is also based on Fuse so it doesn't depend on the kernel. Why would it require FW-supprt?

RMerl, 2019-01-02:
The firmware requires to be aware of which filesystem it's handling.  Each supported filesystem is explicitly managed within rc (and possibly libdisk).  Necessary for determining mount options, mkfs and fsck userspace tools, etc...

Even if it's reverse-engineered, the format is still covered by patents, requiring a licence from Microsoft for legal usage.  I'm not willing to step into a legal minefield especially for something which I consider unnecessary for what is primarily a router, not a NAS.

The firmware doesn't have Fuse support either.  That would be a lot of additional bloat (assuming the kernel version is recent enough to support it - keep in mind most models are still on 2.6.36).

l-g, 2019-01-02:
Whatabout Entware?