RMerl/asuswrt-merlin.ng #183: Advanced DHCP page stops functioning after changing order by + dhcp start ip

dels78, 2018-08-12:
I finally decided to make the jump tonight and I am very happy with the overall improvements... thanks a lot for the great product!

I was browsing all the links in the web interface and all was fine, then I decided to change the start of my DHCP range to help clean up my static reservations and all of a sudden the Advanced_DHCP_Content.asp completely changed behavior (i.e not usable at all) and was not showing my staticlist, even though it was right there in the nvram when I run "nvram get dhcp_staticlist". My first instinct was to try with different browsers, but no dice. I then reverted the change (through ssh since nothing can be done from the page itself), then I rebooted... still no luck.

One thing I noted along the way is that the list in nvram was in a different order than what I had it originally (thank god I still had that old session in buffer!) as I had displayed it ordered by ip (ascending) while looking at the different links and menus.

After much back and forth, I decided to restore the dhcp_staticlist the way it was before and it magically made the webpage behave as expected.

I have attached a screenshot of the defect page, as you can see it is not loading the menu or anything else that is not DHCP Server related, and hitting apply just doesn't do anything.

[image]

RMerl, 2018-08-17:
This indicates something is breaking the page's Javascript.  Open the browser console and look for any error message.  Most common cause is using apostrophes or other illegal characters in the hostname.

dels78, 2018-08-17:
This is correct, only the hostname had been set before I flashed to Merlin, and until I saved something it was displaying fine.

I did however export the nvram dhcp_staticlist to a text file, renamed and re-imported. 

`nvram set dhcp_staticlist="$(cat dhcp_staticlist)" 
nvram commit
`

Thanks for the help

closed 2018-08-17 by dels78