RMerl/asuswrt-merlin.ng #185: AC68U spaming any DNS server

CHazz, 2018-08-12:
hi i want to ask why the router continuously every few seconds spamming the address: dns.msftncsi.com? i do not use this DNS or any other .. i have my own DNS server .. what is it so? and why ? please remove this bug it is very unpleasant and I do not feel good about protecting the privacy of the data stream

skeal1, 2018-08-12:
Try disabling this setting under the tools>other settings tab.  

Wan: Use DNS probes to determine if WAN is up (default: Yes) | Yes No
-- | --

CHazz, 2018-08-13:
thanks i try :)

RMerl, 2018-08-17:
This is by design, this is how Asus checks for the WAN state.  Windows uses the same nameserver to achieve the same result.

closed 2018-08-17 by RMerl

expipiplus1, 2019-10-05:
@skeal1 This setting doesn't appear to be there in firmware version 384.13

EsEnZeT, 2019-11-04:
Found out my way to this issue via my firewall logs & https://discourse.pi-hole.net/t/excessive-requests-for-dns-msftncsi-com/556/25

The only way to stop DNS spam was to **empty** first the:
- Resolve Hostname
- Resolved IP Addresses
under Administration -> System (Network Monitoring) then it can be saved with 'DNS Query' unchecked.