RMerl/asuswrt-merlin.ng #191: RT-AC87U 5GHz Client Don't Appear in Network Map - View List

BlueHornet, 2018-08-22:
In Asuswrt-Merlin release 384.5 my RT-AC87U does not list any of my 5GHz active clients in the "Network Map" -> "View List" in either the "All" or "Interface" Tabs. 

In the "All" tab it lists the Ethernet & 2.4GHz connected clients but none of the active 5GHz clients. In the "Interface" tab it lists the Ethernet & 2.4GHz connected clients in their respective interface type but for the 5GHz type it simply says "No Data in Table" .

Yet if I go to "System Log" -> "Wireless Log" both 2.4GHz and 5GHz clients are displayed in the "List of connected Wireless Clients" according to their respective types.

BlueHornet, 2018-08-27:
> Are you sure? I am running 384.6 and it shows 5GHz devices on the "All" tab.

Yes 100% .I don't see how you are running version 384.6 on a RT-AC87U as this version doesn't exist for this model. I have only yesterday installed 384.7 alpha 2 which has only just been released and it exhibits exactly the same behaviour.

Happy to take screen shots if you like.

The most recent valid versions for the RT-AC87U are 384.5 and 384.7 alpha 2.

DDGo, 2018-08-29:
I use the RTAC88U, so that might be why...

mwolter805, 2018-09-07:
Also experiencing the issue with an AC87R on 384.5 where 5ghz wireless clients do no appear on the network map. Also experiencing issues where 5 ghz clients cannot connect to web servers on the lan. Switching the clients to 2.4 ghz resolves the issue.

duzma, 2018-12-03:
The same issue on my RT-AC87U on firmware 384.7_2 zero clients on 5ghz shown on wireless client list

nfp0, 2019-01-14:
I also have this issue, albeit there are moments where they show up. Sometimes they show up, sometimes they don't. On "System Log" -> "Wireless Log" they always show up consistently.

RMerl, 2019-01-14:
The networkmap code is now closed source, and therefore outside of my control.

closed 2019-01-14 by RMerl