RMerl/asuswrt-merlin.ng #196: Please update OpenSSL to 1.1.0*

iMocW, 2018-09-10:
Mainly for chacha20/chacha20-poly1305 support, should benefits processors without AES instruction.
And since this is NG branch, updating seems legit.

If there's incompatible kit let us know for evaluation.

Originalimoc, 2018-09-10:
Since 1.1.0 and 1.0.2 API are different quite a lot, make it partial support or sideload might be OK?

RMerl, 2018-09-10:
The API change would break all the closed source components from Asus that are linked against openssl 1.0.2.

Which service do you need those ciphers for?

Originalimoc, 2018-09-10:
Oops. Not too necessary actually, on my AC68U HW. Rev.1, AES really capped OpenVPN performance(even it still don't get chacha20 support :/ a related issue tracker opened 2 years now). Seem I should try non-encrypted OpenVPN over an external tunnel.

closed 2018-09-10 by Originalimoc