RMerl/asuswrt-merlin.ng #198: issue with samba

thob, 2018-09-18:
I was using [osmc vero 4k](https://discourse.osmc.tv/t/cant-connect-to-samba-shares-in-my-network-anymore/75029) and had to downgrade Samba there to v1. Otherwise I couldn't see the shares. The vero 4k supports smb v1-3

RMerl, 2018-09-18:
SMB protocol does not dynamically re-negotiate.  After setting the router to V2, make sure to completely reboot your client devices as well.

thob, 2018-09-20:
It works only w/ smbv1. even my Linux Mint 18.3 wasn't able to mount the shares w/ v2

RMerl, 2018-09-23:
Works for me under Ubuntu 18.04:

\\\\192.168.10.1\\Share on /tmp/t type cifs (rw,relatime,**vers=2.0**,cache=strict,username=admin,domain=,uid=0,noforceuid,gid=0,noforcegid,addr=192.168.10.1,file_mode=0755,dir_mode=0755,soft,nounix,serverino,mapposix,rsize=65536,wsize=65536,echo_interval=60,actimeo=1)

Make sure you specify the desired dialect at mount time, i.e.:

```
sudo mount \\\\192.168.10.1\\Share /tmp/t -t cifs -o"username=admin,vers=2.0"
```

thob, 2018-09-23:
ah, the `vers=2.0` makes a difference

closed 2018-09-23 by RMerl