RMerl/asuswrt-merlin.ng #199: "Forward local domain queries to upstream DNS" Setting Ignored in at least 384.6

dreamwraith, 2018-09-19:
I was having a confounding issue with DNS and I think I finally figured it out.

I run my own local DNS Server/Forwarders on my Windows Server. As such, I have the setting "Forward local domain queries to upstream DNS" enabled. My servers also run DNSSEC.

Recently, I noted the radio to enable DNSSEC was not selected on my router config, so I flipped that switch. When I did this, local DNS resolution on the router itself stopped working. It now reports no internet connection, and Dynamic DNS updates do not work for my external DNS provider for my exposed domains. This notwithstanding, traffic still happily works on the remainder of the network.

After initially thinking the problem was with my DNSSEC config on my local servers, I did a bunch of testing and checking, installing and running dig queries from the router itself via ssh.

This is when I discovered the real issue - that the router is, and maybe has for some time, been only using its own localhost address as a dns resolver/forwarder for local dns queries. In essence, it is completely ignoring the "Forward local domain queries to upstream DNS" setting. Regardless of this setting, all locally resolved queries are simply using the localhost somehow. The end result, is that the results are NOT DNSSEC compliant, and do not have the proper DNSSEC segments.

This worked "ok" until I enabled the DNSSEC setting on the router, which exposed the fact that the router itself was not actually respecting the "Forward local domain queries to upstream DNS" setting, and was instead resolving them locally itself.

It does appear, this might be caused primarily only when IPv6 is at play. I note that it is using its link-local IPv6 address on the router to resolve queries locally. Perhaps the "Forward local domain queries to upstream DNS" is being respected, but only for IPv4 queries? And since I run native IPv6, it ends up falling back to the local gateway.

I did note that resolv.conf ONLY lists my IPv4 addresses, and not my IPv6 server addresses. Could that be the problem, that the local resolver is falling back to link-local because it prefers IPv6 first, but no IPv6 addresses are in /tmp/resolv.conf?

dreamwraith, 2018-09-23:
Ok, be that as it may, perhaps there is a separate issue then.

The router is self-resolving dns, rather than using the configured DNS servers. I know this is the case because I tested with dig, both locally on the router, and separately on a debian machine on my network.

The debian machine is properly using my local DNS  to resolve domains, but the router is not - or if it is, it's stripping the dnssec portion of the DNS response somehow.

I'm not knowledgeable enough to know what the cause might be.

So to boil down what I know for absolutely certain: either the router is ignoring configured DNS servers and using something else, OR it's dropping dnssec portion of the DNS response, even when dnssec is enabled. As my isp and my local DNS are both dnssec enabled, even if I use isp DNS I am seeing the same behavior when dnssec setting is enabled on the router.

So perhaps the dnssec setting is the problem? Perhaps enabling dnssec on the router enables the validation check but somehow isn't enabling it to receive the dnssec portion of the packet? Or perhaps there is a bug in core Asus code causing dnssec not to work?

I can resubmit with a better description if you like, I can also provide logs and such if necessary, just let me know what you need.

Thanks

RMerl, 2018-09-23:
> The router is self-resolving dns, rather than using the configured DNS servers. 

This is the desired behaviour, so dnsmasq can handle local name resolution, and also so it can cache responses.  dnsmasq will then use the configured servers for anything it's unable to resolve itself.

You must ensure that your upstream servers do support DNSSEC  Previously, dnsmasq would just silently accept unsigned responses.  Starting with 2.80, unsigned responses are also validated, meaning if it gets an unsigned response while the zone is signed, then it will reject it as it might indicate the response was tinkered in transit, and the signature simply stripped off to prevent validation.  A lot of people got hit by this with 384.6 as they never even realize their upstream servers didn't really support DNSSEC.

Check the dnsmasq.conf config to determine which upstream servers are used for resolution.

dreamwraith, 2018-09-23:
My upstream definitely support dnssec. I've tested with my local servers, as well as Google DNS and my isp Comcast DNS. All support dnssec, and my clients pass resolution test checks verifying dnssec functionality.

Whatever is happening, as soon as I enable dnssec on the router, the router itself refuses to resolve any DNS requests locally, period.

When I run dig via command prompt, it shows it is locally resolving both internal and external addresses, and it shows that it is not receiving dnssec from the local resolution. Every client on my network uses the configured DNS servers, both ipv4 and ipv6, and I tested both my local DNS server, as well as isp and Google, and all clients receive proper dnssec responses from those servers.

So something on the router is either dumping dnssec from the DNS servers and  mangling the DNS packets it receives locally, or it is somehow resolving DNS from some server other than those configured.

To be clear, this isn't even a problem with internal name resolution, it's all name resolution on the router itself. the router will not even resolve google.com when dnssec is enabled on the router.

My concern is that my router itaelf is at risk for DNS poisoning for anything resolved on the router (dynamic DNS calls, entware updates, etc) - my clients are fine regardless since they connect to my local DNS and resolve with dnssec without issue.

closed 2020-09-21 by RMerl