RMerl/asuswrt-merlin.ng #204: iptables filter table FORWARD->other2wan chain missing an item, 384.6

Originalimoc, 2018-09-24:
Since there's ```-A other2wan -i tun+ -j RETURN```
There is a MISSING ```-A other2wan -i tap+ -j RETURN```
Causes WAN connection request from its server side being dropped.

RMerl, 2018-10-09:
I can't reproduce the issue here when using a TAP interface, Internet traffic is going through the tunnel just fine, and the other2wan doesn't list any dropped packets either.  The fact that TAP is a level 2 bridge probably means it's bypassing iptables.

Packet count on the DROP rule is "0".

```
Chain other2wan (1 references)
 pkts bytes target     prot opt in     out     source               destination         
    0     0 RETURN     all  --  tun+   *       0.0.0.0/0            0.0.0.0/0           
    0     0 DROP       all  --  *      *       0.0.0.0/0            0.0.0.0/0
```

Originalimoc, 2018-10-09:
1/ I mean WAN connection request from server side, I found this because certain website can only be access from one of clients, after setting correct route still can't access that website. Since that -j OVPN which -A OVPN -i tapxx -j ACCEPT is after -j other2wan, DROPPED.
2/ Shouldn't tap device means traffic through same bridge(/VLAN(?)) won't go through netfilter/iptables?

closed 2020-09-21 by RMerl