RMerl/asuswrt-merlin.ng #206: mDNS / Bonjour stops working WLAN -> WLAN + LAN after a while

fusionstream, 2018-10-03:
On 384.6 on AC5300, mDNS broadcasts stop getting forwarded after a while (overnight seems to do the trick).

Changing any setting under LAN->IPTV->Special Applications (seemingly) causes various services on the router to restart and the mDNS broadcasts can then be seen again for a period of time. 

Restarting Avahi-daemon has no effect on the missing broadcasts.

Speedliner, 2018-10-10:
maybe related to #162 ?

fusionstream, 2018-10-14:
Hmm. Before I opened this, search didn't show #162 and #102 though it did show #154. So, sorry about opening a new one.

In any case, this seems to have resolved itself in some weird way. I went to toggle settings under LAN->IPTV->Special Applications again to be able to see my devices again and the whole router reset when it came back up (all settings lost). Restored settings from an old backup, made some changes (to bring it up to latest configuration) but did not touch anything under Special Applications.

It's now been more than 12 hours and mDNS/Bonjour seems to still be working. Will reopen if it comes up again.

closed 2018-10-14 by fusionstream

the0ne, 2018-11-08:
I experience the same for #154 after 
1. backing settings
2. restore factory settings
3. restore backed-up settings

However, after several days without the problem, the bug suddenly re-appeared.

fusionstream, 2018-11-09:
My problem came back again a day or so after my last post but then it went away again and hasn't been back since.