RMerl/asuswrt-merlin.ng #209: OpenVPN client watchdog needs to be configurable (needed for 2 factor auth servers)

dledford, 2018-10-05:
When using openvpn client with a server that requires 2 factor auth, the watchdog is useless and does nothing but cause problems.  Several options exist for fixing this problem:

1) Make the watchdog optional.  Have a setting for "What to do when the openvpn client dies: 1) Restart the openvpn client or 2) Set service state to Off and leave dead"
2) Make the watchdog aware of the OpenVPN client exit reason and if it is an authentication failure, then automatically stop reconnect attempts and set the service state to Off

As a side enhancement, when saving a new password, it would be nice if that triggered an automatic change from service state Off to On.  What would be even better, is if there were a setting that allowed you to tell the client that the server uses 2 factor auth, and this option would cause the router to take the password in the password field use it to start the openvpn process via a command line option, but never write the password to nvram or jffs or anywhere else that would persevere past a reboot.

RMerl, 2018-10-09:
The watchdog is just a very simple shell script that runs through a cron job, which checks for the pid.  I recommend creating an openvpn-event script that would delete the cron job once your tunnel is connected.

dledford, 2018-10-10:
Are you suggesting this as the permanent fix or a temporary workaround?

RMerl, 2018-10-10:
For now.  I'm considering adding an option to enable/disable the watchdog, I just want to think about it more carefully first, as this was one of the things I removed a few versions back when I spent a lot of time simplifying OpenVPN configuration by removing seldom-used settings.  So at this time, I haven't made any decision.

dledford, 2018-10-11:
Ok.  I'll just point out that, as in my other issue ticket, the option does break the on/off setting of the service when it won't start up.  Also, there is the option Connection Retry Attempts.  The watchdog renders that particular setting useless.  Maybe turn the watchdog on if that setting is set to -1 (infinite) and off if it has a limit?

RMerl, 2018-10-11:
I'm not convinced there's any need for a watchdog at all either, as OpenVPN itself is fairly robust and very unlikely to flat out crash.  When I found it was broken, I initially completely removed it, before deciding to re-implement it as a basic shell script instead of the original blind restart command it initially used.

dledford, 2018-10-11:
I would find the watchdog much more useful on the server than on the client.  I agree, openvpn is pretty robust and unlikely to crash, but on the server you know it should be restarted in the event of a crash (which is the only reason it should shut down besides being turned off) while on the client you can end up exiting for a number of reasons (auth failure, server offline, server name resolution failure, etc.) that you don't necessarily want to immediately retry the client.

dledford, 2018-10-11:
Although, thinking about it, even on the server, if you have a misconfiguration resulting in a crash, you could get into a restart loop.  I wonder if it isn't worth doing a restart script that is similar to the old SysV init script restart where it would detect a restart loop and shut it down after enough restarts in a certain period of time?

RMerl, 2018-10-11:
The watchdog is only established after the server/client has properly started, so a misconfiguration won't cause any problem.

closed 2018-10-12 by RMerl