RMerl/asuswrt-merlin.ng #212: Wrong configuration for Namecheap DDNS (inadyn)

melchor629, 2018-10-12:
Hi, after the latest update, I saw that the Namecheap DDNS was updating the domain `melchor9000.me` instead of `casita.melchor9000.me` that I was using previously. I tried to look for a solution before posting here so it may help fixing the issue. That's what I found.

The generated configuration looks like this:

```
iterations = 1
custom namecheap {
ddns-server = dynamicdns.park-your-domain.com
ddns-path = "/update?domain=%u&password=%p&host="
hostname = casita
username = "melchor9000.me"
password = "<HIDDEN>"
checkip-command = "/usr/sbin/nvram get wan0_ipaddr"
}
```

Running (via SSH) the command `inadyn -1 -n -l debug`, it shows that the requested URL for the DNS update is `/update?domain=melchor9000.me&password=<HIDDEN>&host=` and that's wrong. Trying to update manually with the following curl command works perfectly.

```
curl -i https://dynamicdns.park-your-domain.com/update\?domain\=melchor9000.me\&password\=<HIDDEN>\&host\=casita
```

Then I decided to look into the code of inadyn, the custom plugin's code. I found [this line][1]. The comment above says that if the `ddns-path` has options, the user is aware of how to put the hostname. So, it's not appending the hostname at the end as the README suggested. Going up into the source code, you will find [this comment][2] that specifies what are the format specifiers, and `%h` is for the hostname.

I tried to run the `inadyn` command again, but now with the `ddns-path` with `/update?domain=%u&password=%p&host=%h` as value (note the `%h` at the end). With that specifier, now it requests the correct URL, and the correct domain is changed. Now IT'S WORKING.

I guess the configuration for Namecheap is wrongly generated and the fix seems to be pretty easy. Nevertheless, as a workaround, I think a postconf script should do the trick meanwhile (sth like this `sed -i "s/&host=\"/\\&host=%h\"/" "$1"`).

Hope it helps to find the issue.

Thank you for your work on this firmware, is really awesome :)

  [1]: https://github.com/troglobit/inadyn/blob/master/plugins/generic.c#L215
  [2]: https://github.com/troglobit/inadyn/blob/master/plugins/generic.c#L85

RMerl, 2018-10-12:
I based the namecheap configuration on the README.md documentation from Inadyn which contained an example configuration, and stated that the hostname was automatically appended to the end of ddns-path.  It's possible then that this does not work as documented, and explicitly specifying the %h parameter should work.

closed 2018-10-12 by RMerl

SimonPilkington, 2018-10-18:
As it turns out, man inadyn.conf states (somewhat indirectly) that hostname is only appended if formatting is not used.

commit a64d848b (2018-10-12): rc: ddns: explicitly append hostname to Namecheap update URL (fixes #212)