RMerl/asuswrt-merlin.ng #213: Why memory management on AC86U is horrible? Crushed many software with no good reason.

Originalimoc, 2018-10-12:
Causes apps just crush under certain circumstances.
Traceroute/dnsmasq and some third party binary just crush due to memory outage. And free -m state 170MiB or more is available.

traceroute -n -I $IP crushed, no issue on AC68U HW. Rev.1

dnsmasq can't handle thousands server=/domain/DNSserver, get random crush, restarted by internal watchdog, have to manually delete some entries to lower the total amount. Not once a problem on AC68U HW. Rev.1

Go library(in kcptun) report crushes due to memory problem when some/a few connections get established. Not applicable to AC68U HW. Rev.1 because it's not profitable to use KCP which requires some processor power (estimate under 10Mbps can be processed under its full CPU load) but CPU in AC86U should get up to around 100Mbps which is certainly profitable.

Again, memory monitor shows NOWHERE near real memory outage. Keep in mind AC68U is with a 256M RAM.

Originalimoc, 2018-10-12:
Further info, kcptun amd64 variant won't cost more than 100MiB in my four running instances started 50 hours ago (12/8/60/100 differ because their load level and GC stuff in GO) but almost no difference at a given time if it's under load(a 300Mbps link just tested) or not. I highly doubt something wrong set about memory management policy in the kernel.

PS.This problem last more than one year when I first noticed it with dnsmasq.
PPS.I don't know why I didn't report it before.😅

RMerl, 2018-10-12:
The HND architecture is entirely under Broadcom's control, there's nothing I can change about it.  The blog module from Broadcom reserves 13% of total available memory for buffers used by their SDK, that's why available memory doesn't match the total memory.

Nothing I can do about it, this is how the architecture has been designed by them.

closed 2018-10-12 by RMerl

Originalimoc, 2018-10-12:
And something like traceroute can crash is "unbelievable" to me, I'm going to reopen this issue later. Did you read this issue carefully?

Originalimoc, 2018-10-12:
Oh no traceroute isn't reporting memory problem, it said no buffer space available.

RMerl, 2018-10-12:
I'm not a kernel developer.  Yes, I know something is odd with the HND plateform, it's mostly visible when spawning multiple processes, but this is in Broadcom's hands, not mine.  When I discussed it with Asus they said it was a limitation in Broadcom's SDK.

Originalimoc, 2018-10-12:
Is traceroute on your control? Something's wrong.

RMerl, 2018-10-12:
You are seeing the symptoms, not the cause.  There's nothing wrong with traceroute itself, it's just a basic busybox applet, and it runs fine here on my own RT-AC86U.

People experienced similar issues spawning multiple curl processes from a shell script as well.  The issue lies at the kernel/SDK level, not at the application level.

Originalimoc, 2018-10-12:
I don't know the bottom cause indeed. I agree the actual problem is not on traceroute. The sendto system call it use doesn't functional properly to my opinion.
So sad this RT-AC86U now became a switch and (a quiet good)Wi-Fi AP to me, I migrated routing part to an external PC. Hope this can be fixed because it costs. And wasted CPU power.
No idea why this is so messed up, especially dnsmasq part, same configuration runs perfect on my AC68U. Moved to PC too.
And why `traceroute -n -I` is it good on yours??? Are you using a PPPoE or what?

Originalimoc, 2018-10-12:
it exits with error `41.157 ms  41.319 mstraceroute: sendto: No buffer space available` with high possibility.

RMerl, 2018-10-12:
Using ICMP packets does result in errors, regular UDP-based doesn't.  No idea why that would make a difference however, unless BCM's internal network buffers on HND have some form of size limit for ICMP packets.

RMerl, 2018-10-12:
Going to open a separate issue for this, since it's unrelated to memory management.

Originalimoc, 2018-10-13:
That traceroute issue last 10 years.... Not going to have a fix Uh.

BTW I found a workaround diminished AC86U's memory problem today, at least works for my use scenario. Use a cheap USB stick to create a swap, since it's NOT actually used so speed is only concerned when creating it(Of course a better one is better, stable wise.) Larger than 256MiB should be enough (I use 1GiB, whatever that 8G stick only cost $4).

Let me know if this will potentially cause any problem.

RMerl, 2018-10-13:
Using a swap file is fine, a lot of users do when using complex adblocking lists or memory-intensive applications such as transmission.