RMerl/asuswrt-merlin.ng #215: CSS font-family definitions don't define fallbacks

jahed, 2018-10-14:
I noticed after switching to GNU/Linux that the web interface's fonts aren't falling back correctly. The most obvious one is the sidebar, which is set to:

```css
.menu {
  font-family: Verdana;
}
```

This means when Verdana isn't available by default it will fall back to the browser's default serif font. Verdana is a sans-serif font, so seeing serif (usually Times New Roman or some variant) makes the interface look odd.

The fix is simple:

```css
.menu {
  font-family: Verdana, sans-serif;
}
```

I can send a PR to fix this across all the stylesheets, but before I do, I'm wondering if style fixes are within the scope of Merlin.

RMerl, 2018-10-17:
Providing a valid fallback class would make sense, however I'm a bit worried by the large amount of changes involved (having seen a lot of CSS definitions lacking a fallback family), which might make things difficult to merge future changes from Asus.

jahed, 2018-10-17:
That's my concern too so I held back on the PR. I don't know what Merlin's merge process is like, nor things like preferred languages (mine is typically Javascript for this scenario) but here's my idea anyway:

I was thinking of writing a script to go through all *.css files, convert them to ASTs, fix any problematic styles and write them back out. Then when a merge from Asus is needed, replace the CSS with the "broken" ones from Asus and re-run the script.

How does that sound?

jahed, 2019-06-24:
Looks like this has been fixed in the latest releases :tada:

closed 2019-06-24 by jahed