RMerl/asuswrt-merlin.ng #221: 384.7.2 - A JavaScript error on "DHCP Server" settings page if there are quotes in the hostnames

tararam, 2018-10-24:
The following line fails on the DHCP Server page (Advanced_DHCP_Content.asp) if there are any single quotes in the hostnames of devices names.

The line 86 failing: 
var dhcp_staticlists = '.....';
The quotes in the dhcp_staticlists value needs to be escaped otherwise JavaScript fails.

RMerl, 2018-10-31:
I've made it more robust by using double quotes instead, however note that a hostname containing a single quote is illegal, and is usually rejected by the webui.

closed 2018-10-31 by RMerl

tararam, 2018-11-07:
Thanks for that, somehow I managed to create couple of those hostnames and cannot change it back to proper as the page fails. Is this change going into next release?

RMerl, 2018-11-07:
Yes.