RMerl/asuswrt-merlin.ng #227: WAN Connection Failed

sergey-prokofiev, 2018-11-18:
Sorry for incomplete report, I'll provide any additional info on request.
Router: ASUS RT-AC86U 
Hardware  version: R1.5

**Description**
- Make factory reset
- Install **384.7_2** asuswrt-merlin firmware using ASUS Firmware Restoration tool
- Changed WAN settings:
  - IP Address -> automatic
  - MAC address ->(hardcoded)
- Connect ISP cable to WAN 
- Configured WiFi
- Reboot

**Result** 
- On ASUS RT-AC86U Web UI: Message 'Provider's DHCP dose not work properly' and in several seconds ->'Disconnected'. Tools->Network->Ethernet Ports->WAN 0->'Unplugged'.
- 'Internet' led on router is red. 
- ISP confirms that there is no connection.
- logs:
  - May  5 05:05:16 WAN_Connection: ISP's DHCP did not function properly.
  - May  5 05:05:16 WAN_Connection: Ethernet link down.

**Workaround**
- Install ASUS 3.0.0.4.384 firmware, everything works.

haveigonemental, 2018-11-23:
I had an identical issue on an RT-AC86U. It would grab an IP from the cable modem just fine and could surf with no problems. Once the 5ghz network came up the WAN would lose its IP and it would show 'Unplugged', same as OP. All of the other behaviors, red LED, etc, were identical as well.

Rolling back the firmware also resolved the issue for me.

crkinard, 2018-11-23:
> Sorry for incomplete report, I'll provide any additional info on request.
> Router: ASUS RT-AC86U
> Hardware version: R1.5
>
> **Description**
>
> * Make factory reset
> * Install **384.7_2** asuswrt-merlin [...]

Did you do a factory reset AFTER flashing the new firmware?

haveigonemental, 2018-11-23:
I did a factory reset after installing the new firmware, the behavior still persisted. After I rolled back the firmware the issue disappeared, I didn't have to anything else to resolve it.

sergey-prokofiev, 2018-11-23:
> @sergey-prokofiev
> 
> Did you do a factory reset AFTER flashing the new firmware? Resetting it before the flash defaults the settings to the previous firmware not the one you are flashing.

Actually I did both(before and after).

z-engine, 2018-12-01:
The same problem :(

kolobock, 2018-12-01:
Setting to No Spoof LAN TTL value should fix the problem. In my case it did

z-engine, 2018-12-02:
This setting turned off for me. I tryed to manipulate with MTU size and DHCP query frequency, but its unsuccessfully.

andrescodder, 2018-12-03:
The problem is with the value of Dhcp lease(ISP server). For example if server(isp) give you 10 min and this time expire,  Asus Merlin try to get new ip address and then wan disconnect. And that problem trigger. I fixed with my ISP router to give 24 hour dhcp reservation ip. Sorry for bad english

closed 2020-09-21 by RMerl