RMerl/asuswrt-merlin.ng #233: Client names everywhere

sectoramen, 2018-12-12:
This is more of an enhancement request. Would it be possible to refer to the client name and use it consistently and not the MAC address everywhere when the client name exists? For instance:

- Malicious site blocking: top client uses names, details uses MAC addresses.
- System Logs/Wireless Logs: Some names are missing, some use the DHCP name (possibly why some names are missing?) not the client name assigned by the user.

RMerl, 2018-12-25:
A lot of these locations (like the Malicious Site Blocking page) are outside of my control.

Wireless Log purposely uses the hostname as known by dnsmasq rather than any description field that might be used by networkmap.  This is by design.

closed 2018-12-25 by RMerl

RMerl, 2018-12-25:
I can at least take care of wireless devices that provide no hostname when obtaining a DHCP lease (those would show with a hostname of "*"), making the Wireless Log page use the networkmap entry (if there is one) instead of showing just "*".