RMerl/asuswrt-merlin.ng #237: more than one host names in Namecheap DDNS

mvadu, 2018-12-18:
After using DD-Wrt for years today I flashed Merlin firmware on ac-87u. I have more than one A records that needs to be updated. In DD-Wrt, I followed the [namecheap guide](https://www.namecheap.com/support/knowledgebase/article.aspx/9356/11/how-to-configure-a-ddwrt-router) to setup the DDNS update script. Trying to use name host names in Merlin firmware did not work.

with `Host Name` set to `@ -a www -a mail -a raspberrypi -a sftp` I get this error. Note the erro shows entirely different screen with trial offer, and blank host name. Same behavior with comma or space separated list of hosts.
[image]

Trying with any one of those host names (one at a time) works just fine.

mvadu, 2018-12-18:
Going through [inadyn readme](https://github.com/troglobit/inadyn) I entered multiple host names as `{"www","mail","raspberrypi","sftp"}` and i got the successful registration. However after refresh the WebUi shows is `{&#34www&#34,&#34mail&#34,&#34raspberrypi&#34,&#34sftp&#34}` which is an eyesore.

RMerl, 2018-12-18:
Multiple hostnames are not supported by the webui, since the nvram containing the hostname has a maximum size limit, and also its content is reused throughout the firmware (for instance to generate the openvpn client config).  Therefore the hostname can only contain one single hostname.

The supported method of using multiple hostnames is to create additional entries using a ddns custom config file.

mvadu, 2018-12-18:
@RMerl okay.. makes sense.. in that case do I leave the WebUI host name blank?

RMerl, 2018-12-18:
I would say configure it with your primary hostname, so it can be used when generating the OpenVPN client config file, and also be used to generate the local cache.  Then, create a custom config to handle all your other aliases.

mvadu, 2018-12-18:
Sure, will do. Btw, going through the inadyn debug trace I just noticed that even with prescribed format, the URL somehow gets messed up `GET /update?domain=adystech.com&password=<passwd>&host=wwwmail` and gives `<ResponseString>No updates; A record not Found;</ResponseString>`. COuld you please check if this is a firmware issue or its upstream repo issue?

themiron, 2018-12-26:
there’s workaround for multiple hostnames - update one dnsomatic account with multiple hostnames on multiple ddns services configured in