RMerl/asuswrt-merlin.ng #238: inputbg.png missing causing all text boxes to have white bkg and white text

samstreak, 2018-12-21:
images/New_ui/inputbg.png does not exist, and is used in many if not almost all text input boxes.  Just started with the upgrade to 384.8_2, but was on 384.7_0 so not 100% sure which one broke it.  I tried ssh/scp a dummy file over with the background color matching a couple of text input boxes that had a specified background color, but the www/images/...  didn't let me write the file with the error that it was a read-only file system.

samstreak, 2018-12-22:
Seems that a reboot fixed whatever issue was causing this.. maybe the .css was cached in the browser or something.

closed 2018-12-22 by samstreak

RMerl, 2018-12-22:
CSS files are cached for a few minutes in your browser, so if you just upgraded, first access might still have an old CSS in cache.  Force-reloading will fix it.