RMerl/asuswrt-merlin.ng #239: TLS verification field field is too small

matfra, 2018-12-22:
Some OpenVPN server (namely all french Freebox Revolution) have a TLS CN that is longer than the 64 char authorized in the field:
Openvpn client 
Verify TLS certificate
Common name

example:
`C=FR, O=Freebox SA, CN=Freebox OpenVPN server <32char hash>
`gets truncated to
`C=FR, O=Freebox SA, CN=Freebox OpenVPN server <18char...
`
thus failing the TLS server verification
A simple workaround is to manually this to the field Custom configuration
`
verify-x509-name "C=FR, O=Freebox SA, CN=Freebox OpenVPN server <32char hash>` in the field

matfra, 2018-12-22:
I believe this is the line that needs to be changed
https://github.com/RMerl/asuswrt-merlin.ng/blob/263dead27428ec8f5ff8979a4d3c52ef7f13adeb/release/src/router/www/Advanced_OpenVPNClient_Content.asp#L1448

RMerl, 2018-12-22:
It's more complicated than that, other locations of the code must also be checked (buffer sizes throughout the code, nvram max size in defaults.c, etc...)

Must also ensure that the nvram size can be increased - Broadcom/Asus enforce some limitations now on newer models, making it difficult (sometimes even impossible) to increase the size limits.

commit c9b72c11 (2018-12-25): openvpn: increase max length of OVPN client common name to 255 chars (closes #239)
closed 2018-12-25 by RMerl