RMerl/asuswrt-merlin.ng #243: Dual-Wan, not able to use IPv6 as destination IP

samip5, 2019-01-15:
Could this be fixed so I can add a policy with IPv6 address as destination?

Image of the screen:
[image]

RMerl, 2019-01-15:
I suspect that interface might possibly only accept IPv4, limiting the max input field length to 255.255.255.255/32.

I don't know if the backend code would support IPv6 either.

samip5, 2019-01-15:
Could you look into it, at least? :)

themiron, 2019-01-15:
right. unfortunately it doesn't support ipv6 at all

samip5, 2019-01-15:
In that case, could there be a IPv6 supported version of such functionality?

RMerl, 2019-01-15:
The Dual WAN routing code is closed sourced and outside of my control.  Best I can do is make sure someone from Asus sees this.

In the meantime, your best chance would be to manually configure things through a firewall-start script.

closed 2019-01-15 by RMerl