RMerl/asuswrt-merlin.ng #245: OpenVPN client auth issues are back (reference issue #209)

dledford, 2019-01-17:
Now, the watchdog script is not the problem, the rc script is the problem.  It used to be that if you enabled the service, then it disconnected (which happens on a semi-regular basis just due to the normal hiccups in traffic over the internet), then the watchdog script would restart the openvpn client.  Now, every two minutes, I see this in the log:

Jan 17 09:26:01 rc_service: service 16774:notify_rc restart_vpnclient1

and it attempts a restart of the service.  This completely locks me out of my company VPN (as well as all other company functions).  I have to reset my 2 factor auth tokens every time this happens, and I'm only allowed to reset them once every 24 hours.  If I get two hiccups in a 24 hour period and I'm not there to stop this runaway retry machine, then I'm done, I can't get any work done after that until I can reset my tokens again.

RMerl, 2019-01-17:
Something must be issuing those restart requests, because rc_service itself does not do anything unless told to.  Make sure you don't have any script or cron job issuing the restart requests.

dledford, 2019-01-17:
Looks like the watchdog crept back in:

admin@vpn-gw-lan:/tmp/etc/openvpn/client1# ls -l
-rw-------    1 admin    root          1540 Jan 17 09:48 ca.crt
-rw-------    1 admin    root           498 Jan 17 09:48 config.ovpn
-rw-------    1 admin    root           193 Jan 17 10:31 status
-rw-------    1 admin    root            26 Jan 17 09:48 up
lrwxrwxrwx    1 admin    root            19 Jan 17 09:48 updown.sh -> /usr/sbin/updown.sh
-rwx------    1 admin    root            78 Jan 17 09:48 vpnc-watchdog1.sh
lrwxrwxrwx    1 admin    root            23 Jan 17 09:48 vpnrouting.sh -> /usr/sbin/vpnrouting.sh
admin@vpn-gw-lan:/tmp/etc/openvpn/client1# more vpnc-watchdog1.sh 
#!/bin/sh
if [ -z $(pidof vpnclient1) ]
then
   service restart_vpnclient1 
fi
admin@vpn-gw-lan:/tmp/etc/openvpn/client1#

RMerl, 2019-01-17:
The watchdog was removed in 384.8, you are running 384.7_2.

dledford, 2019-01-17:
I'm on the latest available for my router :-(  (and I thought I had checked it before and it was gone in my version, but I guess I was wrong)

RMerl, 2019-01-17:
384.8 was not available for the RT-AC87U because the GPL code used was incompatible with the latest RT-AC87U binary blobs available.

For 384.9, I managed to get Asus to provide me with special binary blobs for both the RT-AC3200 and RT-AC87U, so both models should be available for 384.9.  You can experiment with alpha builds published here:

https://asuswrt.lostrealm.ca/test-builds

closed 2019-01-17 by RMerl

dledford, 2019-01-17:
Thanks, I'll try that.