RMerl/asuswrt-merlin.ng #246: VPN Client service state button on config page shows current actual service state, not desired service state

dledford, 2019-01-17:
This is the same as issue #208, which was rendered moot by the resolution to #209.  However, now that issue #209 is back, so it this one.  Essentially, the service On/Off slider on the config page follows the actual service state, not the systems configured and desired service state.  That means that if the service is failing in a repeated loop, even though the service is configured on, the slider will always show off.  That makes it impossible to turn off a service that is restart infinitely.

RMerl, 2019-01-17:
The desired state is not stored in any way, so there's nothing to report on the webui, beside the state of the Start with WAN setting.  This webui portion is working as intended.

closed 2019-01-17 by RMerl

dledford, 2019-01-17:
The watchdog has slipped back in.  If the watchdog goes away entirely again, then this problem goes away.  This is only an issue as long as the watchdog exists.