RMerl/asuswrt-merlin.ng #249: 384.8_2 breaks WAN connection

the0ne, 2019-01-20:
After flashing 384.8_2 my AC88U states "Disconnected".
Solved by stepping back to 384.5

RMerl, 2019-01-20:
Known issue with stock firmware, it's just the detection that fails.  Asus should have it fixed on their end in newer firmware.

the0ne, 2019-01-21:
> Known issue with stock firmware

would have been nice to read a warning in the changelog, but maybe I just missed it.

> it's just the detection that fails.

well, actually no. in fact I actually don't get a connection despite it works with 384.5.

nonekdan, 2019-01-26:
I'm having the same issue with 384.8_2
At first I thought that it was an issue with the fact that I'm using two Sprint Hotspots, but I was doing some work and rerunning my booster cable through the wall and decided to give this another shot.
After a default settings restore I started from scratch and got my USB connected Hotspot going just fine. As soon as I connect the WAN ethernet interface even before fully setting up the connection to the second Hotspot my USB connection starts flapping for some reason.  The Hotspot keeps getting different IP's from SPrint and ultimately I have to turn of the WAN internet connection.  The USB connection settles down and I'm connected to the internet again. 
I am attempting to use Loadbalance by the way as failover really wouldn't do me any good.
I also have a weird thing where my Verizon 4g Repeater can't connect to the Verizon network through my USB connected Sprint Hostpot.  (yeah.. I know)  But if I use a TP-Link extender and extent the Hotspot network and then connect that TP-Link to the Wan port on the asus (no USB) then the Verizon Repeater connects and works just fine.  
The connection is of course much faster with the USB connection so I purchased another hotspot that I was going to loadbalance and then route the Verizon just through the ethernet connected hotspot and all would have been right as rain..  But 10 or so hours later I haven't been able to get them both to connect to the router at the same time.  I didn't want to go back in firmware because I have a smart home and I could not keep my 2.4ghz devices connected prior to this release.  I had to use yet another router ac56 I think it is and just run 2.4ghz so that I could offload them.  I have 60+ devices connected now with 384.8_2 with AC5300 and using smart connect.   I might try to go back to 384.5 but that connection issue with 2.4ghz was killing me.  My switches would lose connectivity and then nothing new could connect without a reboot of the router.  Anyway.. sorry for the ramble.  I'm a bit tired and I get all typy and stuff when that happens..  Piece out.

Geezus42, 2019-01-28:
This appears to be a Duplicate of #227

closed 2019-04-30 by RMerl

the0ne, 2019-05-03:
Hi @RMerl thanks for closing this issue.

Does closing this issue mean that the current version can be flashed onto AC88U without afterwards stating "Disconnected" ?

RMerl, 2019-05-03:
No, it means this issue is far too generic to be an actual bug (in most cases they are caused by misconfiguration of the new WAN monitoring options Asus added in recent releases on the WAN page).  The few remaining cases lie in wanduck, which would be up to Asus to fix.