RMerl/asuswrt-merlin.ng #253: Guest Wireless and Pi-hole or other DNS/DHCP LAN access

sectoramen, 2019-02-04:
Guest wireless clients are unable to access wired Pi-Hole device. Since there is only one place to manually and globally configure DNS, it would make sense to either allow the defined DNS server to be accessed from guest wireless clients even when Access Intranet if set to off, or to have a toggle to allow the defined DNS servers. Thoughts? I haven't tested this, but I would suspect that if DHCP services were turned off on the router so as to use a local DHCP server, this would also cause a problem on the guest network?

jackyaz, 2019-02-07:
Try my script: https://www.snbforums.com/threads/yazfi-enhanced-asuswrt-merlin-guest-wifi-inc-ssid-vpn-client.45924/

sjlehn, 2019-08-20:
I can confirm that guest wireless does not work if you are using the Pi-Hole as your DHCP server - clients will just hang, waiting for an IP address that never comes. There needs to be a way to provide separate DHCP and DNS services for the guest network(s).

sjlehn, 2019-08-20:
I'll have to try @jackyaz's script at some point.

sectoramen, 2019-08-20:
Can a solution be incorporated into the mainstream code?

   - One solution (simpler?) would be to allow all devices to reach the DNS
   servers specified in the configuration settings.
   - A different solution (more work?) may be to add a page on the guest
   wireless settings to specify different DNS servers for guests to point to
   the outside.
   - An awesome solution would be to allow users to manage the firewall
   from the UI by allowing the use of source and destination configurations so
   that specific types of services on the inside can be reached from the guest
   network.

Without a permanent solution, a user is left to not being able to use
PiHole or similar local DNS services at all or having to resort to loading
third-party scripts or use advanced techniques.

RMerl, 2019-08-20:
You either allow LAN access, or you don't.  Pick one, because you can't have both at the same time.  You are basically asking for a security feature to be removed.

This is working as designed, and there are no plans to change it.

closed 2019-08-20 by RMerl

sjlehn, 2019-08-20:
@RMerl is there a way to have the router provide separate DHCP/DNS for the guest network, when there is a custom one being used on the LAN? No need to allow LAN access in that case...

RMerl, 2019-08-20:
You'd have to manually re-implement Guest networks using VLANs instead of just isolating through ebtables.  Doing so is beyond the scope of this project.

jackyaz, 2019-08-20:
YazFi does exactly this....(minus VLANs), but provides separate DHCP scopes, custom DNS and pinhole firewall rules

sectoramen, 2019-08-29:
I am trying to think outside the box here, please follow along and let me know if this will work.

- Set the LAN->DHCP Server->DNS settings to blank so that all clients including guests will used the router as the DNS server.
- Set the router WAN DNS servers to point back inside to the pihole (I m assuming that if the router is used as a DNS server, then it will use these servers)

Is that an option and would traffic flow back? It won't break the security/separation of LAN and guests, which I agree with @RMerl is designed correctly and should not be changed. 

I take it that the switch ports can not be configured individually so as to allow the GUI to individually select a port to be either on LAMN or GUEST? that would allow for guest wired.