RMerl/asuswrt-merlin.ng #258: 384.9 // RT-AC88U - Several errors in USB Application & AiCloud 2.0 webpages

zhuhang-jasper, 2019-02-07:
Firmware: **384.9**
Model: **RT-AC88U** 
HW Ver: **A5**
Browser: **72.0.3626.96 (Official Build) beta (64-bit)**

Errors found in console:
- Uncaught SyntaxError: Invalid or unexpected token **(APP_Installation.asp:77)**
- Uncaught ReferenceError: initial is not defined **(APP_Installation.asp:779)**
- Uncaught SyntaxError: Unexpected token ) **(cloud_main.asp:80)**
- Uncaught ReferenceError: initial is not defined **(cloud_main.asp:619)**

ashokrajar, 2019-04-09:
Same issue on RT-AC3200.

Current firmware : 384.10

zhuhang-jasper, 2019-04-18:
On 384.8 is working fine. That's the reason why I fallback to 384.8

kobudev, 2019-04-19:
Same on RT-AC68U, firmware 384.10_2
Access through IP:port still works

RMerl, 2019-04-30:
Should have been fixed by Asus with 45713 and 5951 GPLs, merged with 384.11.

closed 2019-04-30 by RMerl