RMerl/asuswrt-merlin.ng #259: RT-AC88U - Lowered download speed after creating Guest WiFi

zhuhang-jasper, 2019-02-08:
## Info:
Firmware: **384.9** & **384.8_2**
Model: **RT-AC88U** 
HW Ver: **A5**
Browser: **72.0.3626.96 (Official Build) beta (64-bit)**

## Issues:
After creating a 2.4GHz Guest wifi SSID with bandwidth limit (dl: 50mbps up: 20mbps), 
FW auto changed QoS setting to bandwidth limiter mode (expected), but
Now my download speed is now lowered from **500mbps to 300mbps** (tested using LAN).
I've switched between latest 2 FW versions. Same issue.

After removing the guest WiFi, switching back to Adaptive QoS with automatic bandwidth setting, and rebooting the router, then the speed returns to 500mbps.
I'm suspecting there're some QoS settings that I need to configure?

## QoS Settings:
QoS Type: **Bandwidth Limiter**
Queue Discipline: **sfq**
WAN packet overhead: **none (0)**
Client List: **empty**

RMerl, 2019-02-08:
Bandwidth limiter is incompatible with hardware acceleration, causing it to get disabled.  This is normal behaviour.

closed 2019-02-08 by RMerl

zhuhang-jasper, 2019-02-08:
> Bandwidth limiter is incompatible with hardware acceleration, causing it to get disabled. This is normal behaviour.

So what's the solution to this?

ChriscomIT, 2019-02-12:
@zhuhang-jasper Thats most probably what happens. Whats your CPU usage when you run download speedtest?