RMerl/asuswrt-merlin.ng #260: RT-AC88U - Samba device hidden in LAN connections, but visible in WiFi

zhuhang-jasper, 2019-02-08:
As the title suggests, what is the reason for this behavior?

In Windows 10, I can see samba devices in 'Network' folder (WiFi connection) but not in LAN connection.

zhuhang-jasper, 2019-02-08:
Also happen in some wifi adapters as well. 
I think it is because of some SMB features of the wifi adapter? any idea guys on Windows?

ChriscomIT, 2019-02-12:
Which network adapters (lan/wlan) are u using?

zhuhang-jasper, 2019-02-16:
I'm not sure if its Windows issue or router firmware issue.
At first the network folder is accessible via direct typing "\\RT-AC88U" in explorer, but hidden in the network folder.
Right now, suddenly it is visible in network folder. 
I guess we can close this.

closed 2019-02-16 by zhuhang-jasper