RMerl/asuswrt-merlin.ng #275: Problem to compile the firmware

tuapuikia, 2019-02-17:
Hi,

I try to compile the latest master branch code for AC87U but it always stuck at the following error. It seem like the AC88U merge are caused this issue. May I know how to resolve this issue?

qtn_monitor.c:397:3: warning: implicit declaration of function ‘setAllLedOff_qtn’
qtn_monitor.c:273:1: warning: label ‘QTN_RESET’ defined but not used
 [rc] CC hour_monitor.o
 [rc] CC dnsfilter.o
 [rc] CC natnl_api.o
 [rc] CC erp_monitor.o
 [rc] CC rc
services.o: In function `handle_notifications':
services.c:(.text.handle_notifications+0x31b4): undefined reference to `oauth_google_gen_token_email'
collect2: ld returned 1 exit status
Makefile:656: recipe for target 'rc' failed
make[4]: *** [rc] Error 1

regards,
Stan

RMerl, 2019-02-17:
Make sure you compile from the mainline branch.

vladut123, 2019-02-18:
Ubuntu 18.10 can compile asuswrt-merlin firmware and yes with which packages?

RMerl, 2019-02-18:
The issue tracker is only for development purposes.  If you need technical support or have a question, please use the support forums at https://www.snbforums.com/forums/asuswrt-official.51/ instead.

closed 2019-02-18 by RMerl

tuapuikia, 2019-02-18:
Using merlin branch and it works. CI/CD resume.