RMerl/asuswrt-merlin.ng #277: afraid.freedns.org HTTPS Error on 384.9

wpbullet, 2019-02-23:
I recently upgraded my RT-AC66U that was runniing 380 to an RT-AC5300 running 384.9.

I had afraid DNS running via a custom script, now when I try to use the built-in client in Merlin I get this error in the logs

```
Feb 23 10:30:39 rc_service: watchdog 326:notify_rc start_ddns
Feb 23 10:30:39 start_ddns: update FREEDNS.AFRAID.ORG default@freedns.afraid.org, wan_unit 0
Feb 23 10:30:39 inadyn[2239]: In-a-dyn version 2.5 -- Dynamic DNS update client.
Feb 23 10:30:40 inadyn[2239]: Update forced for alias <subdomain>.chickenkiller.com, new IP# <redacted>
Feb 23 10:30:41 inadyn[2239]: HTTP(S) Transaction failed, error 36: Temporary network error (HTTPS send)
```

I googled and found this issue https://github.com/RMerl/asuswrt-merlin.ng/issues/212 and tried to make the /tmp/etc/inadyn.conf match but the error persists
--

RMerl, 2019-02-23:
Something was wrong with their website recently, I had the same issue last night, I couldn't even access their website using a browser.  It eventually went away.

closed 2019-02-23 by RMerl

wpbullet, 2019-02-23:
I don't think so, when I run the same commands by cURL I don't get any errors related to https or anything

```
curl http://freedns.afraid.org/dynamic/update.php?<hash>=
ERROR: Address 104.x.x.x has not changed.                                                                                                                 
curl https://freedns.afraid.org/dynamic/update.php?<hash>=
ERROR: Address 104.x.x.x has not changed.
```
--

The system log just keeps filling up over and over again every minute making it hard to debug anything else :(. Is there a way to turn it off?

RMerl, 2019-02-23:
The watchdog cannot be disabled, no, you'd have to disable the DDNS service itself, and re-enable it later.

As I said, I had the exact same error last night, and it eventually disappeared.  I just re-tested it here in debug mode, and it's currently working properly for me.

```
inadyn[28430]: Sending IP# update to DDNS server, connecting to freedns.afraid.org([204.140.20.21]:443)
inadyn[28430]: Sending IP# update to DDNS server, initiating HTTPS ...
inadyn[28430]: SSL connection using ECDHE-RSA-AES128-GCM-SHA256
inadyn[28430]: Certificate OK
inadyn[28430]: SSL server cert subject: /OU=Domain Control Validated/OU=EssentialSSL/CN=freedns.afraid.org
inadyn[28430]: SSL server cert issuer: /C=GB/ST=Greater Manchester/L=Salford/O=COMODO CA Limited/CN=COMODO RSA Domain Validation Secure Server CA
inadyn[28430]: Fetching account API key, connecting to freedns.afraid.org([204.140.20.21]:443)
inadyn[28430]: Fetching account API key, initiating HTTPS ...
inadyn[28430]: SSL connection using ECDHE-RSA-AES128-GCM-SHA256
inadyn[28430]: Certificate OK
inadyn[28430]: SSL server cert subject: /OU=Domain Control Validated/OU=EssentialSSL/CN=freedns.afraid.org
inadyn[28430]: SSL server cert issuer: /C=GB/ST=Greater Manchester/L=Salford/O=COMODO CA Limited/CN=COMODO RSA Domain Validation Secure Server CA
inadyn[28430]: Successfully sent HTTPS request!
inadyn[28430]: Successfully received HTTPS response (287 bytes)!
inadyn[28430]: Received API key(s), rc=0:
```

wpbullet, 2019-02-24:
But if the afraid service is working for you and not for me, don't you think that suggests an issue with the router on my end rather than an issue with afraid's servers?

wpbullet, 2019-02-24:
I figured out what this was, the password was wrong! Thank you very much for your time. Now I just have to figure out why OpenVPN isnt working on the external IP and port 443 won't forward.

RMerl, 2019-02-24:
The issue sometimes comes down to routing.  When I was unable to access it two days ago, I was able to access the website from my VPS in the US, but not from my home.