RMerl/asuswrt-merlin.ng #278: OpenVPN not Forwarding from WAN works on LAN

wpbullet, 2019-02-23:
I recently upgraded my RT-AC66U that was runniing 380 to an RT-AC5300 running 384.9, OpenVPN was working on the AC66U.

I have set the OpenVPN configuration to listen on both WAN and LAN. Using LAN works just fine and connects but when I use the external IP address to connect from within my home network it times out for the TLS handshake.

I have port forwarded 1194 from the WAN IP to the router's local IP as well

```
Feb 23 10:38:47 ovpn-server1[988]: 192.168.40.134 TLS: Initial packet from [AF_INET6]::ffff:192.168.40.134:60672, sid=19e38f48 22105410
Feb 23 10:39:47 ovpn-server1[988]: 192.168.40.134 TLS Error: TLS key negotiation failed to occur within 60 seconds (check your network connectivity)
Feb 23 10:39:47 ovpn-server1[988]: 192.168.40.134 TLS Error: TLS handshake failed
Feb 23 10:39:47 ovpn-server1[988]: 192.168.40.134 SIGUSR1[soft,tls-error] received, client-instance restarting
Feb 23 10:39:53 ovpn-server1[988]: 192.168.40.134 TLS: Initial packet from [AF_INET6]::ffff:192.168.40.134:51372, sid=6456342b 9ea01e20
```
--

Checksum for the firmware that was flashed passes as well

wpbullet, 2019-02-24:
I've been digging more on this and I have verified that my ISP (Videotron) does not block port 1194 so it is like the port forwarding on the router is not working at all. When I test from https://www.yougetsignal.com/tools/open-ports/ it shows as closed. Strangely enough, port 32400 does show as open with a very similar port forwarding rule

wpbullet, 2019-02-24:
After disabling the firewall on the router I was able to connect to port 443 which I previously couldn't but OpenVPN port 1194 still isn't open no matter which setting I choose. When I use port forwarding from 1195 to 1194 I can connect via the external IP so it seems something is not quite right with the firewall settings of the OpenVPN server

RMerl, 2019-02-24:
The issue tracker is only for development purposes.  If you need technical support or have a question, please use the support forums at https://www.snbforums.com/forums/asuswrt-official.51/ instead.

closed 2019-02-24 by RMerl

wpbullet, 2019-02-24:
Sorry about that, based on the errors I was seeing it seemed like an error in the firmware that port 1194 wasn't/isn't opening. I will assume that since you have closed this that you have OpenVPN working on the WAN IP

RMerl, 2019-02-24:
That's correct, as well as a few thousand OpenVPN users (I haven't had any other report).