RMerl/asuswrt-merlin.ng #280: There are many exception logs

Alion548, 2019-03-03:
version：AC86U_384.9
a lot of this in the log
`
Mar  3 17:54:43 kernel: nf_conntrack: expectation table full
Mar  3 17:54:43 kernel: nf_conntrack: expectation table full
Mar  3 17:54:43 kernel: nf_conntrack: expectation table full
Mar  3 17:54:43 kernel: nf_conntrack: expectation table full
Mar  3 17:54:43 kernel: nf_conntrack: expectation table full
Mar  3 17:54:43 kernel: nf_conntrack: expectation table full
Mar  3 17:54:43 kernel: nf_conntrack: expectation table full
Mar  3 17:54:43 kernel: nf_conntrack: expectation table full
Mar  3 17:54:43 kernel: nf_conntrack: expectation table full
Mar  3 17:54:43 kernel: nf_conntrack: expectation table full
Mar  3 17:54:48 kernel: net_ratelimit: 58 callbacks suppressed
Mar  3 17:54:48 kernel: nf_conntrack: expectation table full
Mar  3 17:54:48 kernel: nf_conntrack: expectation table full
Mar  3 17:54:48 kernel: nf_conntrack: expectation table full
Mar  3 17:54:48 kernel: nf_conntrack: expectation table full
Mar  3 17:54:48 kernel: nf_conntrack: expectation table full
Mar  3 17:54:48 kernel: nf_conntrack: expectation table full
Mar  3 17:54:48 kernel: nf_conntrack: expectation table full
Mar  3 17:54:48 kernel: nf_conntrack: expectation table full
Mar  3 17:54:48 kernel: nf_conntrack: expectation table full
Mar  3 17:54:48 kernel: nf_conntrack: expectation table full
Mar  3 17:54:55 kernel: net_ratelimit: 6 callbacks suppressed
Mar  3 17:54:55 kernel: nf_conntrack: expectation table full
Mar  3 17:54:56 kernel: nf_conntrack: expectation table full
Mar  3 17:54:56 kernel: nf_conntrack: expectation table full
Mar  3 17:54:56 kernel: nf_conntrack: expectation table full
Mar  3 17:54:56 kernel: nf_conntrack: expectation table full
Mar  3 17:55:01 kernel: nf_conntrack: expectation table full
Mar  3 17:55:01 kernel: nf_conntrack: expectation table full
Mar  3 17:55:01 kernel: nf_conntrack: expectation table full
Mar  3 17:55:04 kernel: nf_conntrack: expectation table full
Mar  3 17:55:04 kernel: nf_conntrack: expectation table full
Mar  3 17:55:05 kernel: nf_conntrack: expectation table full
Mar  3 17:55:05 kernel: nf_conntrack: expectation table full
Mar  3 17:55:06 kernel: nf_conntrack: expectation table full
Mar  3 17:55:06 kernel: nf_conntrack: expectation table full
Mar  3 17:55:15 kernel: nf_conntrack: expectation table full
Mar  3 17:55:15 kernel: nf_conntrack: expectation table full
Mar  3 17:55:15 kernel: nf_conntrack: expectation table full
Mar  3 17:55:15 kernel: nf_conntrack: expectation table full
Mar  3 17:55:19 kernel: nf_conntrack: expectation table full
`

RMerl, 2019-04-04:
Means your router got flooded with connection attempts in a very short period of time, causing its netfilter conntrack table to get filled.

Alion548, 2019-04-04:
[image]
The number of connections is small, and the traffic is not large, it is suspected that the kernel is a bug.

RMerl, 2019-04-04:
It's not.  Google that error message for more details.

Alion548, 2019-04-05:
Netfilter conntrack table is not filled.
``` bash
cat /proc/net/ip_conntrack | wc -l
544

cat /proc/sys/net/ipv4/netfilter/ip_conntrack_max
300000
```

RMerl, 2019-04-19:
conntrack table and expect tables are two different things.

closed 2020-09-21 by RMerl