RMerl/asuswrt-merlin.ng #281: Unable to select language on RT-AC88U on 384.9

Ninj0r, 2019-03-05:
After upgrading from 384.4_2 to to 384.9, the ability to select languages from the drop down menu in the upper right is no longer working.

The router is for the Japanese market and is now locked to the Japanese language.
No other language options are present in the drop down list.

RMerl, 2019-03-05:
Asus now locks the Japanese and Chinese routers to specific languages, and moved the language selection to closed source code to prevent overriding it.  This is working as intended by Asus, and outside of my control.

closed 2019-03-05 by RMerl

Ninj0r, 2019-03-05:
Thank you for the information and update; could you share the version number of the last version that did not have the language lock? I couldn't find it based on the changelog.

Thank you for the fast response and for all your hard work!

RMerl, 2019-03-05:
I don't remember, sorry.

KyonLi, 2019-03-06:
https://www.snbforums.com/threads/firmware-384-21140.47577/#post-416456
The first version that locked languages is 3.0.0.4.384_21140, so maybe 384.6 is the last full language version.

songxinjing, 2019-07-04:
I found this. FYI
https://github.com/RMerl/asuswrt-merlin/commit/567f226a0a317396b5ec5166c512c507421f5d08

songxinjing, 2019-07-04:
I found in file (asuswrt-merlin.ng/release/src/router/httpd/web.c) these codes below was losted.
so outside of control, sigh~
```
if(!strncmp(nvram_safe_get("territory_code"), "JP", 2) && strcmp(nvram_safe_get(ATE_FACTORY_MODE_STR()), "1")){
		websWrite(wp, "<li style=\"visibility:hidden;\"><dl><a href=\"#\"><dt id=\"selected_lang\"></dt></a>\\n");
	}
```