RMerl/asuswrt-merlin.ng #282: 384.9_0 // RT-AC87U - Problem creating directories on a Samba share (USB Application)

Wick3DRebel, 2019-03-07:
Since updated to latest Release firmware (384.9_0), for RT-AC87U, I've been having problems copying files to the USB device attached to the router).
Using Windows 10: the drives mounts correctly but when attempting to copy over a whole directory, containing multiple files and other directories, Windows presents me with "Error 0x8007010B: The directory name is invalid". This seems to occur with directory creation only, but the directories are still created on the SMB share.
I've tried enabled "SMB 1.0/CIFS File Sharing Support" under "Turn Windows features on or off" to no avail. Cycling the SMB protocol between 1/2/1+2 also did not resolve the issue.
I've reverted back to 384.7_2 and the issue is gone.

RMerl, 2019-03-07:
Issue possibly introduced by an updated Tuxera driver in recent GPLs:

https://www.snbforums.com/threads/error-0x8007010b-on-rt-ax88u.54205/

I need to investigate to make sure that the documented workaround does not introduce different issues.

zhuhang-jasper, 2019-03-15:
This also happens in 384.8_2

RMerl, 2019-03-15:
Fixed with 62ee83024224cc1c625291bea84f434afc33ebf1 .

closed 2019-03-15 by RMerl