RMerl/asuswrt-merlin.ng #283: Kernel panic

jmperez0, 2019-03-09:
I am running RT-AC68U with version 384.8_2 and I got in the log the following message:

```
Mar  9 20:56:17 kernel: kernel BUG at mm/slub.c:2837!
Mar  9 20:56:17 kernel: Unable to handle kernel NULL pointer dereference at virtual address 00000000
Mar  9 20:56:17 kernel: pgd = cb9a4000
Mar  9 20:56:18 kernel: [00000000] *pgd=9be68031, *pte=00000000, *ppte=00000000
Mar  9 20:56:18 kernel: Internal error: Oops: 817 [#1] PREEMPT SMP
Mar  9 20:56:18 kernel: last sysfs file: /sys/class/net/vlan1/brport/port_no
Mar  9 20:56:18 kernel: module:  tun	 bf7fd000	 12274
Mar  9 20:56:18 kernel: module:  nfsd	 bf7e1000	 81901
Mar  9 20:56:18 kernel: module:  exportfs	 bf7db000	 2959
Mar  9 20:56:18 kernel: module:  lockd	 bf7c5000	 58157
Mar  9 20:56:18 kernel: module:  sunrpc	 bf791000	 161418
Mar  9 20:56:18 kernel: module:  xt_set	 bf78a000	 7395
Mar  9 20:56:18 kernel: module:  ip_set	 bf77e000	 23781
Mar  9 20:56:18 kernel: module:  nf_nat_sip	 bf777000	 5031
Mar  9 20:56:18 kernel: module:  nf_conntrack_sip	 bf76e000	 15713
Mar  9 20:56:18 kernel: module:  nf_nat_h323	 bf767000	 4761
Mar  9 20:56:18 kernel: module:  nf_conntrack_h323	 bf759000	 33807
Mar  9 20:56:18 kernel: module:  nf_nat_rtsp	 bf753000	 3202
Mar  9 20:56:18 kernel: module:  nf_conntrack_rtsp	 bf74d000	 4067
Mar  9 20:56:18 kernel: module:  nf_nat_ftp	 bf747000	 1144
Mar  9 20:56:18 kernel: module:  nf_conntrack_ftp	 bf740000	 4909
Mar  9 20:56:18 kernel: module:  ip6table_mangle	 bf73a000	 934
Mar  9 20:56:18 kernel: module:  sr_mod	 bf732000	 10842
Mar  9 20:56:18 kernel: module:  ipt_account	 bf72a000	 8380
Mar  9 20:56:18 kernel: module:  cdrom	 bf71d000	 30901
Mar  9 20:56:18 kernel: module:  usblp	 bf715000	 10321
Mar  9 20:56:18 kernel: module:  thfsplus	 bf6f9000	 81917
Mar  9 20:56:18 kernel: module:  tntfs	 bf671000	 498661
Mar  9 20:56:18 kernel: module:  tfat	 bf638000	 189126
Mar  9 20:56:18 kernel: module:  ext2	 bf624000	 52776
Mar  9 20:56:18 kernel: module:  ext4	 bf5e3000	 221674
Mar  9 20:56:18 kernel: module:  crc16	 bf5dd000	 1007
Mar  9 20:56:18 kernel: module:  jbd2	 bf5ca000	 48993
Mar  9 20:56:18 kernel: module:  ext3	 bf5a8000	 106409
Mar  9 20:56:18 kernel: module:  jbd	 bf597000	 42367
Mar  9 20:56:18 kernel: module:  mbcache	 bf590000	 4599
Mar  9 20:56:18 kernel: module:  usb_storage	 bf581000	 34290
Mar  9 20:56:18 kernel: module:  sg	 bf576000	 19855
Mar  9 20:56:18 kernel: module:  sd_mod	 bf56a000	 21983
Mar  9 20:56:18 kernel: module:  scsi_wait_scan	 bf564000	 416
Mar  9 20:56:18 kernel: module:  scsi_mod	 bf53f000	 108738
Mar  9 20:56:18 kernel: module:  cdc_mbim	 bf539000	 3137
Mar  9 20:56:18 kernel: module:  qmi_wwan	 bf532000	 5816
Mar  9 20:56:18 kernel: module:  cdc_wdm	 bf52b000	 7252
Mar  9 20:56:18 kernel: module:  cdc_ncm	 bf523000	 8762
Mar  9 20:56:18 kernel: module:  rndis_host	 bf51c000	 5260
Mar  9 20:56:18 kernel: module:  cdc_ether	 bf516000	 4036
Mar  9 20:56:18 kernel: module:  asix	 bf50e000	 10856
Mar  9 20:56:18 kernel: module:  cdc_acm	 bf505000	 12614
Mar  9 20:56:18 kernel: module:  usbnet	 bf4fc000	 11900
Mar  9 20:56:18 kernel: module:  mii	 bf4f6000	 3367
Mar  9 20:56:18 kernel: module:  ip6t_LOG	 bf4ef000	 4494
Mar  9 20:56:18 kernel: module:  ip6table_filter	 bf4e9000	 750
Mar  9 20:56:18 kernel: module:  ohci_hcd	 bf4df000	 17926
Mar  9 20:56:18 kernel: module:  ehci_hcd	 bf4d1000	 32382
Mar  9 20:56:18 kernel: module:  usbcore	 bf4ad000	 102572
Mar  9 20:56:18 kernel: module:  jffs2	 bf48f000	 91274
Mar  9 20:56:18 kernel: module:  zlib_deflate	 bf485000	 19489
Mar  9 20:56:18 kernel: module:  nf_nat_pptp	 bf47f000	 1602
Mar  9 20:56:18 kernel: module:  nf_conntrack_pptp	 bf479000	 3355
Mar  9 20:56:18 kernel: module:  nf_nat_proto_gre	 bf473000	 887
Mar  9 20:56:18 kernel: module:  nf_conntrack_proto_gre	 bf46d000	 3296
Mar  9 20:56:18 kernel: module:  wl	 bf02f000	 4196700
Mar  9 20:56:18 kernel: module:  dpsta	 bf028000	 5900
Mar  9 20:56:18 kernel: module:  et	 bf011000	 64476
Mar  9 20:56:18 kernel: module:  igs	 bf009000	 11887
Mar  9 20:56:18 kernel: module:  emf	 bf000000	 15225
Mar  9 20:56:18 kernel: Modules linked in: tun nfsd exportfs lockd sunrpc xt_set ip_set nf_nat_sip nf_conntrack_sip nf_nat_h323 nf_conntrack_h323 nf_nat_rtsp nf_conntrack_rtsp nf_nat_ftp nf_conntrack_ftp ip6table_mangle sr_mod ipt_account cdrom usblp thfsplus tntfs(P) tfat(P) ext2 ext4 crc16 jbd2 ext3 jbd mbcache usb_storage sg sd_mod scsi_wait_scan scsi_mod cdc_mbim qmi_wwan cdc_wdm cdc_ncm rndis_host cdc_ether asix cdc_acm usbnet mii ip6t_LOG ip6table_filter ohci_hcd ehci_hcd usbcore jffs2 zlib_deflat
Mar  9 20:56:18 kernel: CPU: 0    Tainted: P             (2.6.36.4brcmarm #1)
Mar  9 20:56:18 kernel: PC is at __bug+0x1c/0x28
Mar  9 20:56:18 kernel: LR is at __bug+0x18/0x28
Mar  9 20:56:18 kernel: pc : [<c004113c>]    lr : [<c0041138>]    psr: 60000013
Mar  9 20:56:18 kernel: sp : cb881d70  ip : 00000000  fp : 000005ea
Mar  9 20:56:18 kernel: r10: cf5a9040  r9 : 000005ea  r8 : 00000000
Mar  9 20:56:18 kernel: r7 : cf3f0e00  r6 : cf3f0e28  r5 : cf3f0e00  r4 : 00000000
Mar  9 20:56:18 kernel: r3 : 00000000  r2 : 00000000  r1 : cde93600  r0 : 00000024
Mar  9 20:56:18 kernel: Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
Mar  9 20:56:18 kernel: Control: 10c53c7d  Table: 9b9a404a  DAC: 00000015
Mar  9 20:56:18 kernel: Process iptraf (pid: 6105, stack limit = 0xcb880270)
Mar  9 20:56:18 kernel: Stack: (0xcb881d70 to 0xcb882000)
Mar  9 20:56:18 kernel: 1d60:                                     9a000000 c0377ce4 00000000 cf3f0e00
Mar  9 20:56:18 kernel: 1d80: cf3f0e28 cf3f0e00 00000000 c037a918 cb99dc00 cb881f5c cf3f0e28 cf3f0e00
Mar  9 20:56:18 kernel: 1da0: 00000000 c01f3df4 cb99dc00 c02b35c4 00000014 00000000 00000000 00000000
Mar  9 20:56:18 kernel: 1dc0: 00001805 00000000 00000000 000005ea 00000000 00000000 00004500 cb881f5c
Mar  9 20:56:18 kernel: 1de0: c02eae68 00000000 00000000 00000000 be8483b8 c01e8604 00000000 00000000
Mar  9 20:56:18 kernel: 1e00: 0000000a 00000004 00000000 00000001 ffffffff 00000000 00000000 00000000
Mar  9 20:56:18 kernel: 1e20: 00000000 00000000 cb84e400 00000000 00000000 00000000 00000000 00000008
Mar  9 20:56:18 kernel: 1e40: cb881e78 00000000 cb881e80 00002050 3b9ac9ff cf950600 00000001 cf950600
Mar  9 20:56:18 kernel: 1e60: c8225f80 00000c00 010a1642 00000000 c0415630 00193f6e 2e59ba1c c041fabc
Mar  9 20:56:18 kernel: 1e80: 00000000 00004500 cf5a9040 be8481e0 00000000 cb881f5c cb881ea8 c01fbb18
Mar  9 20:56:18 kernel: 1ea0: cb881edc be84b4d0 00307262 00000000 0000000c 00000c00 cb881f84 00004500
Mar  9 20:56:18 kernel: 1ec0: cf5a9040 cb881edc be84b4d0 be8483b8 cb880000 c01ea8dc 2e59ba1c 00080011
Mar  9 20:56:18 kernel: 1ee0: 00000008 06040001 317e1640 006b8859 00000000 cb8280a0 be8481e0 be8481e0
Mar  9 20:56:18 kernel: 1f00: 00000003 c003dc68 a37c05ef 13294dd9 a37c05ef cb880000 be8481d8 00000001
Mar  9 20:56:18 kernel: 1f20: cb881f48 be8481d8 00000008 00000000 00000001 c00d93a8 00000000 02fad893
Mar  9 20:56:18 kernel: 1f40: 00000000 0000c349 00000000 02fad893 be848200 be8481d8 be848200 cb881edc
Mar  9 20:56:18 kernel: 1f60: 00000012 cb881f78 00000001 00000000 00000000 0000004e be84baba 00003f16
Mar  9 20:56:18 kernel: 1f80: 00000000 fffffff7 00002050 be8483b8 be8481d4 00000004 00000124 c003dc68
Mar  9 20:56:18 kernel: 1fa0: 00000000 c003dac0 be8483b8 be8481d4 00000003 be84b4d0 00004500 00000000
Mar  9 20:56:18 kernel: 1fc0: be8483b8 be8481d4 00000004 00000124 00000008 00000003 be848380 be8483b8
Mar  9 20:56:18 kernel: 1fe0: be8481c0 be8481b4 00016358 40214978 60000010 00000003 2079614d 30203520
Mar  9 20:56:18 kernel: [<c004113c>] (PC is at __bug+0x1c/0x28)
Mar  9 20:56:18 kernel: [<c004113c>] (__bug+0x1c/0x28) from [<c0377ce4>] (kfree+0xf0/0xfc)
Mar  9 20:56:18 kernel: [<c0377ce4>] (kfree+0xf0/0xfc) from [<c037a918>] (__kfree_skb+0x28/0x1dc)
Mar  9 20:56:18 kernel: [<c037a918>] (__kfree_skb+0x28/0x1dc) from [<c01f3df4>] (skb_free_datagram+0x10/0x38)
Mar  9 20:56:18 kernel: [<c01f3df4>] (skb_free_datagram+0x10/0x38) from [<c02b35c4>] (packet_recvmsg+0x378/0x42c)
Mar  9 20:56:18 kernel: [<c02b35c4>] (packet_recvmsg+0x378/0x42c) from [<c01e8604>] (sock_recvmsg+0xa0/0xb4)
Mar  9 20:56:18 kernel: [<c01e8604>] (sock_recvmsg+0xa0/0xb4) from [<c01ea8dc>] (sys_recvfrom+0x88/0xe8)
Mar  9 20:56:18 kernel: [<c01ea8dc>] (sys_recvfrom+0x88/0xe8) from [<c003dac0>] (ret_fast_syscall+0x0/0x30)
Mar  9 20:56:18 kernel: Code: e59f0010 e1a01003 eb0a2c7f e3a03000 (e5833000) 
Mar  9 20:56:18 kernel: ---[ end trace c02348b3275f06bf ]---
```
The router then rebooted by itself.

johnbarratt, 2019-04-15:
Just had the same thing happen on an RT-AC3200 running the latest firmware, 384.10_2. Also note that the router was factory reset and manually reconfigured after firmware upgrade. Previously I had stability issues that caused the router to lock up instead of reset like this, but I was unable to capture any error details until now.

The router is running as a PPPoE gateway with all three WIFI bands enabled, QoS disabled, a few virtual servers configured, DHCP server enabled with some static hosts, SNMP enabled, remote logging enabled.

```May  5 16:04:59 kernel: _ Reboot message ... _______________________________________________________
May  5 16:04:59 kernel: <1>Unable to handle kernel NULL pointer derefere�U�at virtual address 00000000
May  5 16:04:59 kernel: <1>pgd = 80004000
May  5 16:04:59 kernel: <1>[00000000] *pgd=00000000
May  5 16:04:59 kernel: <0>Internal error: Oops: 817 [#1] PREEMPT SMP
May  5 16:04:59 kernel: <0>last sysfs file: /sys/module/nf_conntrack/parameters/hashsize
May  5 16:04:59 kernel: <4>module:  nf_nat_sip	 7f3f0000	 5031
May  5 16:04:59 kernel: <4>module:  nf_conntrack_sip	 7f3e7000	 15713
May  5 16:04:59 kernel: <4>module:  nf_nat_h323	 7f3e0000	 4761
May  5 16:04:59 kernel: <4>module:  nf_conntrack_h323	 7f3d2000	 33807
May  5 16:04:59 kernel: <4>module:  nf_nat_rtsp	 7f3cc000	 3202
May  5 16:04:59 kernel: <4>module:  nf_conntrack_rtsp	 7f3c6000	 4067
May  5 16:04:59 kernel: <4>module:  nf_nat_ftp	 7f3c0000	 1144
May  5 16:04:59 kernel: <4>module:  nf_conntrack_ftp	 7f3b9000	 4909
May  5 16:04:59 kernel: <4>module:  ip6table_mangle	 7f3b3000	 934
May  5 16:04:59 kernel: <4>modmodule:  thfsplus	 7f38e000	 85357
May  5 16:04:59 kernel: <4>module:  tntfs	 7f30e000	 468922
May  5 16:04:59 kernel: <4>module:  tfat	 7f2d5000	 191259
May  5 16:04:59 kernel: <4>module:  ext2	 7f2c0000	 53788
May  5 16:04:59 kernel: <4>module:  ext4	 7f27f000	 222306
May  5 16:04:59 kernel: <4>module:  crc16	 7f279000	 1007
May  5 16:04:59 kernel: <4>module:  jbd2	 7f265000	 49553
May  5 16:04:59 kernel: <4>module:  ext3	 7f242000	 106589
May  5 16:04:59 kernel: <4>module:  jbd	 7f231000	 42799
May  5 16:04:59 kernel: <4>module:  mbcache	 7f22a000	 4599
May  5 16:04:59 kernel: <4>module:  usb_storage	 7f21b000	 34410
May  5 16:04:59 kernel: <4>module:  sg	 7f210000	 20039
May  5 16:04:59 kernel: <4>module:  sd_mod	 7f204000	 22167
May  5 16:04:59 kernel: <4>module:  scsi_wait_scan	 7f1fe000	 416
May  5 16:04:59 kernel: <4>module:  scle:  ohci_hcd	 7f1cf000	 17918
May  5 16:04:59 kernel: <4>module:  ehci_hcd	 7f1c2000	 31653
May  5 16:04:59 kernel: <4>module:  xhci_hcd	 7f1af 7f1a9000	 3129
May  5 16:04:59 kernel: <4>module:  qmi_wwan	 7f1a2000	 5780
May  5 16:04:59 kernel: <4>module:  cdc_wdm	 7f19b000	 7252
May  5 16:04:59 kernel: <4>module:  cdc_ncm	 7f193000	 8750
May  5 16:04:59 kernel: <4>module:  rndis_host	 7f18c000	 4936
May  5 16:04:59 kernel: <4>module:  cdc_ether	 7f186000	 3187
May  5 16:04:59 kernel: <4>module:  asix	 7f17e000	 10832
May  5 16:04:59 kernel: <4>module:  cdc_acm	 7f176000	 10357
May  5 16:04:59 kernel: <4>module:  usbnet	 7f16d000	 11161
May  5 16:04:59 kernel: <4>module:  mii	 7f167000	 3367
May  5 16:04:59 kernel: <4>module:  usbcore	 7f144000	 102094
May  5 16:04:59 kernel: <4>module:  ip6t_LOG	 7f13d000	 4494
May  5 16:04:59 kernel: <4>module:  ip6table_filter	 7f137000	 750
May  5 16:04:59 kernel: <4>module:  jffs2	 7f119000	 91538
May  5 16:04:59 kernel: <4>module:  zlib_de:  dhd	 7f02e000	 242573
May  5 16:04:59 kernel: <4>module:  et	 7f01b000	 48925
May  5 16:04:59 kernel: <4>module:  igs	 7f013000	 11927
May  5 16:04:59 kernel: <4>module:  emf	 7f00a000	 15345
May  5 16:04:59 kernel: <4>module:  ctf	 7f000000	 20115
May  5 16:04:59 kernel: <4>Modules linked in: nf_nat_sip nfnntrack_h323 nf_nat_rtsp nf_conntrack_rtsp nf_nat_ftp nf_conntrack_ftp ip6table_mangle usblp thfsplus tntfs(P) tfat(P) ext2 ext4 crc16 jbd2 ext3 jbd mbcache usb_storage sg sd_mod scsi_wait_scan scsi_mod ohci_hcd ehci_hcd xhcx cdc_acm usbnet mii usbcore ip6 ctf(P) [last unloaded: tdts]
May  5 16:04:59 kernel: <4>CPU: 1    Tainted: P             (2.6.36.4brcmarm #1)
May  5 16:04:59 kernel: <4>PC is at dhdpcie_bus_process_mailbox_i  psr: 60000113
May  5 16:04:59 kernel: <4>sp : 8f83ded0  ip : 7f0553ec  fp : 00000000
May  5 16:04:59 kernel: <4>r10: 8004d1d4  r9 : 80479f20  r8 : 8047ecd8
May  5 16:04:59 kernel: <4>r7 : 7f05dae0  r6 : 00000000  r5 : 00000100  r4 : 8eb2f000
May  5 16:04:59 kernel: <4>r3 : 00000000  r2 : 0000000a  r1 : 00000000  r0 : 00000834
May  5 16:04:59 kernel: <4>Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM53c7d  Table: 8df2804a  DAC: 00000017
May  5 16:04:59 kernel: <0>Process swapper (pid: 0, stack limit = 0x8f83c270)
May  5 16:04:59 kernel: <0>S              00000000 8eb2f000 00000000 10000000
May  5 16:04:59 kernel: <0>dee0: 8f83c000 8eb2f000 00000100 00000000 88 8007a31c 00000001
May  5 16:04:59 kernel: <0>df20: 80408058 8f83c000 00000103 8004d148a8b0
May  5 16:04:59 kernel: <0>df40: 90810000 80068b34 00000006 0000000a 00000008 8f83c000 8047ecd8 8004d9d0
May  5 16:04:59 kernel: <0>df60: 8f83c000 00000000 8f83c000 0000001f 00000000 8007abcc 000000a3 803ca440
May  5 16:04:59 kernel: <0>df80: ffffffff b010010c9f68 00000003 8d0e9500
May  5 16:04:59 kernel: <0>dfa0:b0a8 80412b80 8042b214 80000000 413fc090
May  5 16:04:59 kernel: <0>dfc0: 0000001f 000000: 80050bc4 80050d88 8f84806a 0000001f 10c03c7d 80008148 23514641 c600250d
[... cut ...]
May  5 16:04:59 kernel: <4>[<800574f0>] (do_page_fault+0x150/0x1ec) from [<8004f3a4>] (do_DataAbor_DataAbort+0x30/0x9c) from [<803c9f0c>] (__dabt_svc+0x4c/0x60)
May  5 16:04:59 kernel: 4>Exception stack(0x8f83de88 to  80479f20 8004d1d4 00000000  ^C803c9f0c>] (__dabt_svc+0x4c/0x60_process_mailbox_intr+0x164/0x1f0 [dhd])
May  5 16:04:59 kernel: <4>[<7f041870>] (dhdpcie_bus_process_mailbox_intr+0x164
May  5 16:04:59 kernel: ____________________________________________________________________________
```

aselvan, 2019-11-16:
@johnbarratt 
Just curious, were you able to find the root-cause of this problem? I have the same exact problem w/ my RT-AC3200 on ASUS firmware (3.0.0.4.382_51640-g679a7e3). I don't remember when this random reboots problem started to happen but definitely it was not the case before; the router would run weeks and months w/ out any issues, but now reboots randomly multiple times in a week and sometimes multiple times during the day.

jmperez0, 2019-11-16:
Hi, I think I was running iptraf and perf at the same time. I assumed that there might be a problem related to network monitoring, and did not attempt to do it again. In the web UI I have the statistics section of the traffic analyzer disabled. Perhaps you could check if disabling it in your router makes it more stable.
Cheers.

aselvan, 2019-11-16:
> Hi, I think I was running iptraf and perf at the same time. I assumed that there might be a problem related to network monitoring, and did not attempt to do it again. In the web UI I have the [...]
I am running traffic analyzer and air-protection, turned off traffic analyzer to see if that helps...although both of these were running fine well over a year for me with no problem.

closed 2020-09-21 by RMerl