RMerl/asuswrt-merlin.ng #289: OpenVPN unable to connect on 384.10

TootToot168, 2019-03-26:
Model: RT-AC86U

Exact same OpenVPN connection works well on v384.9

Log from v384.10

> Mar 26 13:06:09 ovpn-client1[1823]: OpenVPN 2.4.7 arm-buildroot-linux-gnueabi [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD] built on Mar 24 2019
Mar 26 13:06:09 ovpn-client1[1823]: library versions: OpenSSL 1.1.1b  26 Feb 2019, LZO 2.08
Mar 26 13:06:09 ovpn-client1[1824]: WARNING: No server certificate verification method has been enabled.  See http://openvpn.net/howto.html#mitm for more info.
Mar 26 13:06:09 ovpn-client1[1824]: NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
Mar 26 13:06:09 ovpn-client1[1824]: TCP/UDP: Preserving recently used remote address: [AF_INET]206.123.155.4:53
Mar 26 13:06:09 ovpn-client1[1824]: UDP link local: (not bound)
Mar 26 13:06:09 ovpn-client1[1824]: UDP link remote: [AF_INET]206.123.155.4:53
Mar 26 13:06:10 ovpn-client1[1824]: WARNING: this configuration may cache passwords in memory -- use the auth-nocache option to prevent this
Mar 26 13:06:10 ovpn-client1[1824]: VERIFY ERROR: depth=0, error=CA signature digest algorithm too weak: C=HK, ST=HK, L=HongKong, O=PureVPN, OU=IT, CN=PureVPN, name=PureVPN, emailAddress=mail@host.domain
Mar 26 13:06:10 ovpn-client1[1824]: OpenSSL: error:1416F086:lib(20):func(367):reason(134)
Mar 26 13:06:10 ovpn-client1[1824]: TLS_ERROR: BIO read tls_read_plaintext error
Mar 26 13:06:10 ovpn-client1[1824]: TLS Error: TLS object -> incoming plaintext read error
Mar 26 13:06:10 ovpn-client1[1824]: TLS Error: TLS handshake failed
Mar 26 13:06:10 ovpn-client1[1824]: SIGUSR1[soft,tls-error] received, process restarting
Mar 26 13:06:15 ovpn-client1[1824]: WARNING: No server certificate verification method has been enabled.  See http://openvpn.net/howto.html#mitm for more info.
Mar 26 13:06:15 ovpn-client1[1824]: NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
Mar 26 13:06:15 ovpn-client1[1824]: TCP/UDP: Preserving recently used remote address: [AF_INET]206.123.155.4:53
Mar 26 13:06:15 ovpn-client1[1824]: UDP link local: (not bound)
Mar 26 13:06:15 ovpn-client1[1824]: UDP link remote: [AF_INET]206.123.155.4:53
Mar 26 13:06:15 ovpn-client1[1824]: VERIFY ERROR: depth=0, error=CA signature digest algorithm too weak: C=HK, ST=HK, L=HongKong, O=PureVPN, OU=IT, CN=PureVPN, name=PureVPN, emailAddress=mail@host.domain
Mar 26 13:06:15 ovpn-client1[1824]: OpenSSL: error:1416F086:lib(20):func(367):reason(134)
Mar 26 13:06:15 ovpn-client1[1824]: TLS_ERROR: BIO read tls_read_plaintext error
Mar 26 13:06:15 ovpn-client1[1824]: TLS Error: TLS object -> incoming plaintext read error
Mar 26 13:06:15 ovpn-client1[1824]: TLS Error: TLS handshake failed
Mar 26 13:06:15 ovpn-client1[1824]: SIGUSR1[soft,tls-error] received, process restarting
Mar 26 13:06:20 ovpn-client1[1824]: WARNING: No server certificate verification method has been enabled.  See http://openvpn.net/howto.html#mitm for more info.
Mar 26 13:06:20 ovpn-client1[1824]: NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
Mar 26 13:06:20 ovpn-client1[1824]: TCP/UDP: Preserving recently used remote address: [AF_INET]206.123.155.4:53
Mar 26 13:06:20 ovpn-client1[1824]: UDP link local: (not bound)
Mar 26 13:06:20 ovpn-client1[1824]: UDP link remote: [AF_INET]206.123.155.4:53
Mar 26 13:06:21 ovpn-client1[1824]: VERIFY ERROR: depth=0, error=CA signature digest algorithm too weak: C=HK, ST=HK, L=HongKong, O=PureVPN, OU=IT, CN=PureVPN, name=PureVPN, emailAddress=mail@host.domain
Mar 26 13:06:21 ovpn-client1[1824]: OpenSSL: error:1416F086:lib(20):func(367):reason(134)
Mar 26 13:06:21 ovpn-client1[1824]: TLS_ERROR: BIO read tls_read_plaintext error
Mar 26 13:06:21 ovpn-client1[1824]: TLS Error: TLS object -> incoming plaintext read error
Mar 26 13:06:21 ovpn-client1[1824]: TLS Error: TLS handshake failed
Mar 26 13:06:21 ovpn-client1[1824]: SIGUSR1[soft,tls-error] received, process restarting
Mar 26 13:06:26 ovpn-client1[1824]: WARNING: No server certificate verification method has been enabled.  See http://openvpn.net/howto.html#mitm for more info.
Mar 26 13:06:26 ovpn-client1[1824]: NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
Mar 26 13:06:26 ovpn-client1[1824]: TCP/UDP: Preserving recently used remote address: [AF_INET]206.123.155.4:53
Mar 26 13:06:26 ovpn-client1[1824]: UDP link local: (not bound)
Mar 26 13:06:26 ovpn-client1[1824]: UDP link remote: [AF_INET]206.123.155.4:53
Mar 26 13:06:26 ovpn-client1[1824]: VERIFY ERROR: depth=0, error=CA signature digest algorithm too weak: C=HK, ST=HK, L=HongKong, O=PureVPN, OU=IT, CN=PureVPN, name=PureVPN, emailAddress=mail@host.domain
Mar 26 13:06:26 ovpn-client1[1824]: OpenSSL: error:1416F086:lib(20):func(367):reason(134)
Mar 26 13:06:26 ovpn-client1[1824]: TLS_ERROR: BIO read tls_read_plaintext error
Mar 26 13:06:26 ovpn-client1[1824]: TLS Error: TLS object -> incoming plaintext read error
Mar 26 13:06:26 ovpn-client1[1824]: TLS Error: TLS handshake failed
Mar 26 13:06:26 ovpn-client1[1824]: SIGUSR1[soft,tls-error] received, process restarting
Mar 26 13:06:31 ovpn-client1[1824]: WARNING: No server certificate verification method has been enabled.  See http://openvpn.net/howto.html#mitm for more info.
Mar 26 13:06:31 ovpn-client1[1824]: NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
Mar 26 13:06:31 ovpn-client1[1824]: TCP/UDP: Preserving recently used remote address: [AF_INET]206.123.155.4:53
Mar 26 13:06:31 ovpn-client1[1824]: UDP link local: (not bound)
Mar 26 13:06:31 ovpn-client1[1824]: UDP link remote: [AF_INET]206.123.155.4:53
Mar 26 13:06:32 ovpn-client1[1824]: VERIFY ERROR: depth=0, error=CA signature digest algorithm too weak: C=HK, ST=HK, L=HongKong, O=PureVPN, OU=IT, CN=PureVPN, name=PureVPN, emailAddress=mail@host.domain
Mar 26 13:06:32 ovpn-client1[1824]: OpenSSL: error:1416F086:lib(20):func(367):reason(134)
Mar 26 13:06:32 ovpn-client1[1824]: TLS_ERROR: BIO read tls_read_plaintext error
Mar 26 13:06:32 ovpn-client1[1824]: TLS Error: TLS object -> incoming plaintext read error
Mar 26 13:06:32 ovpn-client1[1824]: TLS Error: TLS handshake failed
Mar 26 13:06:32 ovpn-client1[1824]: SIGUSR1[soft,tls-error] received, process restarting
Mar 26 13:06:42 ovpn-client1[1824]: WARNING: No server certificate verification method has been enabled.  See http://openvpn.net/howto.html#mitm for more info.
Mar 26 13:06:42 ovpn-client1[1824]: NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
Mar 26 13:06:42 ovpn-client1[1824]: TCP/UDP: Preserving recently used remote address: [AF_INET]206.123.155.4:53
Mar 26 13:06:42 ovpn-client1[1824]: UDP link local: (not bound)
Mar 26 13:06:42 ovpn-client1[1824]: UDP link remote: [AF_INET]206.123.155.4:53
Mar 26 13:06:42 ovpn-client1[1824]: VERIFY ERROR: depth=0, error=CA signature digest algorithm too weak: C=HK, ST=HK, L=HongKong, O=PureVPN, OU=IT, CN=PureVPN, name=PureVPN, emailAddress=mail@host.domain
[... 3 lines cut ...]
Mar 26 13:06:42 ovpn-client1[1824]: TLS Error: TLS handshake failed
Mar 26 13:06:42 ovpn-client1[1824]: SIGUSR1[soft,tls-error] received, process restarting
Mar 26 13:07:02 ovpn-client1[1824]: WARNING: No server certificate verification method has been enabled.  See http://openvpn.net/howto.html#mitm for more info.
Mar 26 13:07:02 ovpn-client1[1824]: NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
Mar 26 13:07:02 ovpn-client1[1824]: TCP/UDP: Preserving recently used remote address: [AF_INET]206.123.155.4:53
Mar 26 13:07:02 ovpn-client1[1824]: UDP link local: (not bound)
Mar 26 13:07:02 ovpn-client1[1824]: UDP link remote: [AF_INET]206.123.155.4:53
Mar 26 13:07:03 ovpn-client1[1824]: VERIFY ERROR: depth=0, error=CA signature digest algorithm too weak: C=HK, ST=HK, L=HongKong, O=PureVPN, OU=IT, CN=PureVPN, name=PureVPN, emailAddress=mail@host.domain
Mar 26 13:07:03 ovpn-client1[1824]: OpenSSL: error:1416F086:lib(20):func(367):reason(134)
Mar 26 13:07:03 ovpn-client1[1824]: TLS_ERROR: BIO read tls_read_plaintext error
Mar 26 13:07:03 ovpn-client1[1824]: TLS Error: TLS object -> incoming plaintext read error
Mar 26 13:07:03 ovpn-client1[1824]: TLS Error: TLS handshake failed
Mar 26 13:07:03 ovpn-client1[1824]: SIGUSR1[soft,tls-error] received, process restarting
[... cut ...]
Mar 26 13:07:43 ovpn-client1[1824]: VERIFY ERROR: depth=0, error=CA signature digest algorithm too weak: C=HK, ST=HK, L=HongKong, O=PureVPN, OU=IT, CN=PureVPN, name=PureVPN, emailAddress=mail@host.domain
Mar 26 13:07:43 ovpn-client1[1824]: OpenSSL: error:1416F086:lib(20):func(367):reason(134)
Mar 26 13:07:43 ovpn-client1[1824]: TLS_ERROR: BIO read tls_read_plaintext error
Mar 26 13:07:43 ovpn-client1[1824]: TLS Error: TLS object -> incoming plaintext read error
Mar 26 13:07:43 ovpn-client1[1824]: TLS Error: TLS handshake failed
Mar 26 13:07:43 ovpn-client1[1824]: SIGUSR1[soft,tls-error] received, process restarting
Mar 26 13:08:12 rc_service: httpd 767:notify_rc stop_vpnclient1

RMerl, 2019-03-26:
```
Mar 26 13:06:10 ovpn-client1[1824]: VERIFY ERROR: depth=0, error=CA signature digest algorithm too weak: C=HK, ST=HK, L=HongKong, O=PureVPN, OU=IT, CN=PureVPN, name=PureVPN, emailAddress=mail@host.domain
```

Ask your provider if they have a newer CA for you to use, the one provided uses a weak (insecure) digest.

closed 2019-03-26 by TootToot168

RMerl, 2019-03-26:
Just to further document this: OpenSSL 1.1.x no longer accepts certificates using a weak digest.  PureVPN's old CA certificate was signed with SHA1, which is considered too weak.  PureVPN's latest config files should contain a certificate using a more secure signature.