RMerl/asuswrt-merlin.ng #291: Upgrade notification won't go away after upgrading to 384.10 (AC88U)

DDGo, 2019-03-26:
Pretty much title. I get a notification about needing to upgrade despite being on the most recent 384.10 release. This happened right after upgrading from .6 I believe. Checking the forum [shows](https://www.snbforums.com/threads/release-asuswrt-merlin-384-10-is-now-available.55742/page-3#post-474579) me that atleast one other person has the same issue. If there is any info you need, please let me know.

RMerl, 2019-03-26:
Try disabling the Beta notification under Tools -> Other Settings.

I suspect this is a side-effect of Asus recently removing support for the beta channels, which might have impacted my own code.  I am considering removing the Beta channel as well from the update notification with 384.11.

DDGo, 2019-03-26:
@RMerl I already had that and the new version check disabled. I pretty much always update manually, but lately I've noticed that despite having the update check disabled it will still search for updates regularly.

vladut123, 2019-03-26:
@RMerl when 384.11 will be available?

RMerl, 2019-03-26:
Patience, 384.10 is not even one week old...

vladut123, 2019-03-26:
@rmerl when asuswrt-merlin 384.11 will be available

RMerl, 2019-03-26:
It does not mean it's searching for updates (because if it was, it would have actually update the state).  You can manually clear the state over ssh:

```
nvram set webs_state_flag=0
nvram commit
```

DDGo, 2019-03-26:
Thanks for the workaround in mean time. Really appreciate that and the fact that you keep this firmware up to date

master3395, 2019-04-11:
This happen to me to. @RMerl 
Thanks for the workaround, worked for me to 👍

RMerl, 2019-04-30:
Should have been taken care of with the changes done with 384.10_2 and 384.11 to the firmware update handling.

closed 2019-04-30 by RMerl