RMerl/asuswrt-merlin.ng #292: Port Forwarding configuration missing after upgrade to 384.10 AC68U

accelle17, 2019-03-29:
After upgrading from 384.4 to 384.10 last 3/28, I have lost the port forwarding configurations. I tried adding by 'Add Profile' but after refresh or changing to different menu, it seems it's not saving.
Enable port forwarding is set to on.

This is the only logs after adding new profile:

> Mar 29 17:49:40 rc_service: httpd 20386:notify_rc restart_firewall
> Mar 29 17:49:41 miniupnpd[23104]: shutting down MiniUPnPd
> Mar 29 17:49:41 nat: apply nat rules (/tmp/nat_rules_eth0_eth0)
> Mar 29 17:49:42 miniupnpd[23659]: HTTP listening on port 46400
> Mar 29 17:49:42 miniupnpd[23659]: Listening for NAT-PMP/PCP traffic on port 5351

RMerl, 2019-03-29:
Your router is probably too low on free nvram - check under Tools -> Sysinfo.

Closing this as this is not a firmware bug.  Please use the support forums at SNBForums if you need further assistance.

closed 2019-03-29 by RMerl

accelle17, 2019-03-30:
Thanks for clue. I have freed up some by removing old vpn configs. May need to backup and restore to default in order to clear stale nvram space.