RMerl/asuswrt-merlin.ng #294: 349.10 and 349.10_4 wan bandwidth problems

PabloC, 2019-04-06:
I did a simple speed test with 349.10 and 349.10_4 and found a bandwidth problem.

Maximum speed is 150/200MB of downstream. It should be 600MB (I'am based in Spain, with a Movistar fiber optic connection which is 600/600MB simetric connection).

I've used Movistar Test de Velocidad, Fast.es and Speedtest.net to test the speed and in all the tests, the maximum speed is 150/200MB as max. All the tests were done connected to an ethernet cable of course.

After downgrading to 349.9, the problem is fixed, and I'm getting 600/600 OK.

My router is an Asus Wireless RT-AC68U.

Do you need any information to help on the debug of this? Let me know.

powerlebrato, 2019-04-13:
The same problem (but with different firmwares)
Spain, Movistar 600/600, AC68U. With firm 384.10_2 Speedtest (and Movistar test) results in ±150/200 Mbps. Returning to 384.9_0 the result is ±630/615 Mbps.

Voidx, 2019-04-17:
+1 with orange.

Voidx, 2019-04-19:
some update about this problem?

RMerl, 2019-04-19:
Please use the support forums if you need help troubleshooting performance issues with your connection.

closed 2019-04-19 by RMerl