RMerl/asuswrt-merlin.ng #295: [REQUEST] minidlna root_container setting

Keltere, 2019-04-14:
Hi, i wanted to know if it's possible to add an additional setting for `root_container` value.

From minidlna config file:
```
# use different container as root of the tree
# possible values:
#  + "." - use standard container (this is the default)
#  + "B" - "Browse Directory"
#  + "M" - "Music"
#  + "V" - "Video"
#  + "P" - "Pictures"
# if you specify "B" and client device is audio-only then "Music/Folders" will be used as root
```

kolobock, 2019-04-14:
you may use ‘minidlna.postconf’ to replace default config to your own. 
```
root@asuswrt:/jffs/scripts# cat minidlna.postconf 
#!/bin/sh

CONF=$1
CUSTOMCONF=/jffs/configs/minidlna.conf.base

mv $CONF $CONF.system

log "Installing custom $CONF"
cp $CUSTOMCONF $CONF
grep "serial=" $CONF.system >> $CONF
grep "model_number=" $CONF.system >> $CONF
grep "uuid=" $CONF.system >> $CONF
```

Keltere, 2019-04-14:
> you may use ‘minidlna.postconf’ to replace default config to your own.
Thanks i've already did with the custom appending config file `/jffs/configs/minidlna.conf.add` with just `root_container=B`
 
Just because it's a useful setting i just asked if this could be implemented on the interface.

closed 2020-09-21 by RMerl