RMerl/asuswrt-merlin.ng #304: client on wifi instantly disconnects when cisco anyconnect starts

JeffGiroux, 2019-04-25:
Was running 384.9 and then upgraded to 384.10_2. Same issue. Client computer (windows) has Cisco Anyconnect. Wifi connection (no VPN) works great. As soon as Cisco Anyconnect establishes a connection to VPN server, Wifi gets booted.

Need help knowing where to look.

I have searched many different internet articles and they lead to being "asus issues". Then people switch vendors and/or Merlin to ddwrt. I'm trying to avoid that.

odkrys, 2019-04-25:
You can try openconnect-gui which is compatible with anyconnect.
There are various reports about anyconnect stability issue on win10 wifi.

JeffGiroux, 2019-04-25:
We messed with some settings last night. In the end, it just took a quick jolt. Meaning, I hooked computer to ethernet cable first, then established VPN with Cisco Anyconnect. This kept the connection open. I then disconnected ethernet, then connected Wifi again. Then connected Cisco Anyconnect and it gave a quick "disconnect" icon to the wifi again but then stayed connected. 

It now works (on my router at least) each time as of this "ethernet workaround" test last night. Again, all I did was simply connect to VPN while on ethernet. Didn't install anything or do anything. Then I swapped back to wifi and things are working fine.

closed 2019-04-25 by JeffGiroux