RMerl/asuswrt-merlin.ng #307: Jumbo frames

multiflexi, 2019-04-27:
I use AC68U and even though jumbo packets are enabled in administration, ifconfig on the device and ping to the device show that they are not, MTU remains 1500. It is possible to maualy increase MTU of eth1 and eth2 but not the virtual br0 device.

`br0       Link encap:Ethernet  HWaddr 78:24:AF:E6:77:30  
          inet addr:192.168.2.1  Bcast:192.168.2.255  Mask:255.255.255.0
          UP BROADCAST RUNNING ALLMULTI MULTICAST  MTU:1500  Metric:1
          RX packets:12536 errors:0 dropped:0 overruns:0 frame:0
          TX packets:12864 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:2035683 (1.9 MiB)  TX bytes:5628777 (5.3 MiB)

eth1      Link encap:Ethernet  HWaddr 78:24:AF:E6:77:30  
          UP BROADCAST RUNNING ALLMULTI MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:84675
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:163 

eth2      Link encap:Ethernet  HWaddr 78:24:AF:E6:77:34  
          UP BROADCAST RUNNING ALLMULTI MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:80541
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:169 

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          UP LOOPBACK RUNNING MULTICAST  MTU:16436  Metric:1
          RX packets:1307 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1307 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:235672 (230.1 KiB)  TX bytes:235672 (230.1 KiB)

tun21     Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  
          inet addr:192.168.100.1  P-t-P:192.168.100.1  Mask:255.255.255.0
          UP POINTOPOINT RUNNING NOARP PROMISC MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

vlan1     Link encap:Ethernet  HWaddr 78:24:AF:E6:77:30  
          UP BROADCAST RUNNING ALLMULTI MULTICAST  MTU:1500  Metric:1
          RX packets:17361 errors:0 dropped:0 overruns:0 frame:0
          TX packets:13465 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:2640787 (2.5 MiB)  TX bytes:5729498 (5.4 MiB)

wl0.1     Link encap:Ethernet  HWaddr 78:24:AF:E6:77:31  
          UP BROADCAST RUNNING ALLMULTI MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:84675
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

wl1.1     Link encap:Ethernet  HWaddr 78:24:AF:E6:77:35  
          UP BROADCAST RUNNING ALLMULTI MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:80541
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
`

RMerl, 2019-04-27:
The issue tracker is only for development purposes.  If you need technical support or have a question, please use the support forums at https://www.snbforums.com/forums/asuswrt-official.51/ instead.

closed 2019-04-27 by RMerl

multiflexi, 2019-04-28:
I don't have a question, I think this is a bug. The change of settings in web interface does not work.

RMerl, 2019-04-28:
Sorry, I thought you were asking how to increase the MTU values.

The jumbo frame handling is part of Broadcom's proprietary code, so I don't know how it's supposed to be implemented.  Enable_jumbo_frame(void) runs a low-level command to configure the switch at the register levels.  Only Asus/Broadcom would know more about how this is supposed to be implemented.

BillFleming, 2020-10-04:
FYI I see the same lack of JumboFrame support on the RT-AX88U for anyone that is wondering:

ASUSWRT-Merlin RT-AX88U 384.19_0 Fri Aug 14 19:20:07 UTC 2020
@RT-AX88U-A2F0:/tmp/home/root# **ifconfig**
br0       Link encap:Ethernet  HWaddr 0C:9D:92:03:A2:F0  
          inet addr:192.168.....
          UP BROADCAST RUNNING ALLMULTI MULTICAST  MTU:1500  Metric:1
          RX packets:41699 errors:0 dropped:16 overruns:0 frame:0
          TX packets:53933 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:5105891 (4.8 MiB)  TX bytes:46857374 (44.6 MiB)

eth0      Link encap:Ethernet  HWaddr 0C:9D:92:03:A2:F0  
          inet addr:.....
          UP BROADCAST RUNNING ALLMULTI MULTICAST  MTU:**1500**  Metric:1
          RX packets:37705 errors:0 dropped:0 overruns:0 frame:0
          TX packets:25388 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:38399482 (36.6 MiB)  TX bytes:3442824 (3.2 MiB)