RMerl/asuswrt-merlin.ng #310: 384.11_alpha4-g2169d263b9 - internet connection status = Disconnected

master3395, 2019-04-30:
[image]

master3395, 2019-05-01:
Update: After around 12 hours, the internet goes down.
And you have to restart the router, in order to get the internet back.

andrescodder, 2019-05-02:
That happend When router (wan)try to get a New dhcp  address. A temporal Fix is what in your isp modem put dhcp léase to 24 hours o More

facaldoggi, 2019-05-06:
Same here with **RT-AC86U on 384.10_2** - WAN disconnects randomly stating (like in the pic) internet status disconnected with red LED.
Did a factory reset, but it did not help. it's unclear to me if the issue was introduced with 384.10 or 384.9....

**I had to go back to 384.8_2.**

chrisgross, 2019-05-12:
My RT-AC68U has this problem on 384.11, too. Had to downgrade to 384.10 to get WAN back.

SR4ven, 2019-05-13:
I'm having a similar issue since 384.11 where after a few hours my downlink is capped to below 10Mbit/s, sometimes below 1Mbit/s (I have got a 200Mbit/s connection).
I'm sure it's not my modem or ISP as I can perform a speedtest directly on my modem too and the speed is perfect there. The modem is in bridge mode so it doesn't run a DHCP server or anything.
The router doesn't completely lose connection as you all described but I believe my modem goes into some kind of emergency mode to "force" a restart/DHCP renew. Restarting the RT-AC87U doesn't help, even worse, it cuts the connection completely and isn't able to reconnect. For some reason I need to restart the modem to reset it all back to a working state.

master3395, 2019-06-15:
I tried updating to "384.12_alpha2-g6a46c21cce", but the same issue here.
It gets disconnected after 24 hours ish.
And then you can disconnect the device from the router, and connect it again, and then the internet works well, but it's as @SR4ven says, it stays around 10Mbit/s down and 5 down.
Tested on 5 devices.
And connecting to the ISP's hardware router, makes it go 500 V 500.
So the router is slowing it REALLY down.

SR4ven, 2019-06-15:
@master3395 have you tried [resetting to factory defaults](https://www.snbforums.com/threads/faq-nvram-and-factory-default-reset.22822/)?
It might have fixed my issues, but I can't really tell as I'm not using VPN/QoS and the TrendMicro features anymore. I'm on 384.11_2 now

master3395, 2019-06-15:
> @master3395 have you tried [resetting to factory defaults](https://www.snbforums.com/threads/faq-nvram-and-factory-default-reset.22822/)?
> It might have fixed my issues, but I can't really tell as I'm not using VPN/QoS and the TrendMicro features anymore. I'm on 384.11_2 now

Haven't done NVRAM reset, but I've done a factory reset.
Just before i updated to the version i got now.
I do use VPN and QoS.

asaddon, 2019-06-15:
@master3395 Doing a factory reset DOES clear NVRAM.

asaddon, 2019-06-15:
And it's recommend to do factory reset AFTER you're on the latest firmware.

chrisgross, 2019-06-28:
FYI, I'm not sure if my problem was exactly the same as everyone elses's here, but it must have been fixed in 384.12 because I just upgraded with no problems. No factory reset required.

master3395, 2019-06-29:
Haven't seen this issue in 384.12_beta2-g7e33ba641a.

closed 2019-06-29 by master3395