RMerl/asuswrt-merlin.ng #312: Add WSD discovery for SMB

mohd-akram, 2019-05-14:
This will allow Windows 10 to discover the SMB share.

See:
https://github.com/christgau/wsdd
https://github.com/Andy2244/wsdd2
https://github.com/kochinc/wsdd2

RMerl, 2019-05-14:
First one is a no-go as it requires Python, but the two other ones are worth investigating to see what their requirements are.

mohd-akram, 2019-05-15:
I managed to compile and run the Andy2244/wsdd2 one on my RT-AC88U and it worked. I finally saw the computer icon in Network.

To compile:

    CC=/opt/brcm/bin/arm-brcm-linux-uclibcgnueabi-gcc LD_LIBRARY_PATH=/opt/brcm/lib make

I had to copy `ifaddrs.c` from `src/router/shared` and modify the Makefile to add it.

Resource usage from `top`:

```
  PID  PPID USER     STAT   VSZ %VSZ CPU %CPU COMMAND
24342 24253 admin    S     1012  0.2   1  0.0 ./wsdd2
```

`ls`:

    -rwxrwxr-x    1 admin    root       56.6K May 15 07:39 wsdd2

RMerl, 2019-05-15:
I was able to cross-compile it as well as part of the build enviro itself, however I had no luck having shares appear on my test laptop.  I didn't have much time to spend troubleshooting it tho.

mohd-akram, 2019-05-15:
You might have to start the `Function Discovery Provider Host` service, although it was already started in my case. Also, enable `Turn on network discovery` in `Manage advanced sharing settings` (search from Start menu).

RMerl, 2019-05-15:
Both services were already running, and network discovery was also enabled.  However I ran my tests on the RT-AX88U, which has a slightly different environment (glibc instead of uclibc, and a newer kernel).  The test laptop runs an Insider build however, so it might have a few "quirks".

I'll spend more time on it later when I'm free.

RMerl, 2019-05-15:
Also Windows does detect other devices through WSD already (if you set the Network listing to Details, you can then add a "Discovery Method" column to that window)

mohd-akram, 2019-05-15:
Thanks for the tip. It was actually using NetBIOS somehow. I found the issue - wsdd2 reads `/etc/machine-id` which should contain a ["hexadecimal, 32-character, lowercase ID"](https://www.freedesktop.org/software/systemd/man/machine-id.html). After creating that file on the router, the discovery method changes to WSD.

RMerl, 2019-05-16:
In my case it was failing because I generated a machine-id through uuidgen, which is a different format.

wsd sure could use some better error handling.

mohd-akram, 2019-05-16:
I did the exact same thing first :)

I removed some lines in the `#define DEBUG` macro in `wsdd.h` so it would print everything. I later discovered the help (`-h`) which has some interesting options.

RMerl, 2019-05-16:
Its use of testparam will also need to be addressed, as Asuswrt-Merlin doesn't ship with it.

RMerl, 2019-05-16:
aae_deviceid cannot be relied upon, as it implies the router should have a working AsusNAT service.

I am considering generating an md5 hash our of the router's LAN's MAC address, which should be unique, and easily doable as md5sum is already part of the firmware.

mohd-akram, 2019-05-16:
Sounds good. I can't think of any other unique ID. Slightly tweaked, to get rid of the filename (hyphen in this case) and run as daemon:

	echo $(nvram get lan_hwaddr) | md5sum | cut -b -32 > /etc/machine-id
	wsdd2 -d

RMerl, 2019-05-16:
To compensate for lack of testparam, I am considering implementing two new command line arguments to provide the Workgroup and the Netbios name.  First one can be retrieved from the st_samba_workgroup nvram value, while Netbios name will have to be generated through libdisk (which takes the computer_name, nvram value and ensures its adjusted to fit within the netbios requirements).

mohd-akram, 2019-05-16:
I did a quick test with a "testparm" alternative which might be sufficient:

```shell
#!/bin/sh
for arg; do case $arg in
	--parameter-name=*)
		pattern="/^[[:blank:]]*${arg#*=}[[:blank:]]*=[[:blank:]]*/"
		sed -n "s$pattern/p" /etc/smb.conf | tail -n 1
		exit
esac; done
```

RMerl, 2019-05-16:
Or as a built-in replacement (example code):

```
#include <string.h>
#include <stdio.h>

int main(){
        char netbios[18], workgroup[64];
        char buf[256];
        FILE *fp;

        strcpy(netbios,"");
        strcpy(workgroup,"");

        if (!(fp = fopen("/tmp/smb.conf","r"))) {
                printf("Cannot find smb config\n");
                return -1;
        }

         while (fgets(buf, sizeof(buf), fp) && (!*netbios || !*workgroup)) {
                if (!*workgroup)
                        sscanf(buf, "workgroup = %63[^\n]s", workgroup);
                if (!*netbios)
                        sscanf(buf, "netbios name = %17[^\n]s", netbios);
        }
        fclose(fp);

        if (*netbios && *workgroup)
                printf("WG = %s, NB = %s.\n", workgroup,netbios);
        else
                printf("Not found!\n");
}
```

RMerl, 2019-05-16:
Andy2244's patch for OpenWRT looks less clunky than hashing the MAC to generate a machine-id:

https://github.com/Andy2244/openwrt-extra/blob/master/wsdd2/patches/001-add_uuid_boot_id.patch

RMerl, 2019-05-16:
wsdd2 implemented through the following commits:

b22e061684 wsdd: generate ID from the kernel's boot-id instead of hashing the MAC (patch from OpenWRT)
c1ab39dce7 rc: implement wsdd stop/start alongside Samba
80971f456b wsdd: implement Asus-specific code, enable build recipe
d03c883c50 wsdd: add build recipes
c4ca06e988 wsdd: initial commit

closed 2019-05-16 by RMerl

mohd-akram, 2019-05-16:
Awesome, thanks!