RMerl/asuswrt-merlin.ng #314: Intercept NTP client requests

keboard, 2019-05-24:
Running version 384.11_2 on an RT-AC86U and with the local NTP server option set to 'yes' and the **intercept NTP client requests** set to **'no'**, local subnet clients were indeed using the router as their ntp server.   Among other devices, there are four debian based machines on my lan; two are running with ntpd v.4.2.8p10 and two with v4.2.6p6.  The two v.4.2.8p10 systems were time serving the network using ntp.org pools as their source.  The v4.2.6p6 systems had ntp.conf files as follows and pointing to the two servers...

```
driftfile /var/lib/ntp/ntp.drift
restrict default -4 nomodify nopeer noquery notrap
restrict default -6 nomodify nopeer noquery notrap   # if your machine has IPv6 connectivity
restrict 224.0.1.1 mask 255.255.255.255 nomodify     # if using multicast (most folks are not)
restrict 127.0.0.1
restrict ::1
restrict 192.168.28.0 mask 255.255.255.0 
server 192.168.28.183 iburst prefer
server 192.168.28.184 iburst prefer
server 127.127.1.0
fudge 127.127.1.0 stratum 10
```

After a while, ntpq -pn showed the router (x.x.28.7) being used

```
     remote           refid      st t when poll reach   delay   offset  jitter
================================================================================
+192.168.28.183   198.50.238.163   3 u   17   64   77    0.528   33.728   0.713
*192.168.28.184   192.168.28.7      3 u   16   64   77    0.631    1.033   0.280
 127.127.1.0     .LOCL.          10 l  160   64   74    0.000    0.000   0.001
+192.168.28.7     71.80.83.0       2 u   25   64   77    0.334    0.835   0.341
```

The lan ntp server machines (x.x.28.183 and .184) had similar ntp.conf files but instead of having the two server lines (.183 and .184), there were two pool entries using ntp.org pool machines.   ntpq -pn on the lan ntp servers also showed the router providing times; in fact in the output above, the .184 ntp server machine was using .7 (the router) as it's source.

It appears NTP client requests are getting intercepted even with the radio button set to 'no'.

crkinard, 2019-05-24:
Check /etc/dnsmasq.conf.

See if there is a setting 'dhcp-option=lan,4,<routerIP>'.

keboard, 2019-05-24:
Thanks for that.  Here are the dhcp-option lines from the router's dnsmasq.conf.  There isn't the Time Server '4' param, but that 'ntp-server' line looks suspect.  (x.x.28.7 being the router.)

```
dhcp-option=lan,3,192.168.28.7
dhcp-option=lan,15,xyz
dhcp-option=lan,44,192.168.28.7
dhcp-option=lan,252,"\n"
dhcp-option=option:ntp-server,192.168.28.7
```

crkinard, 2019-05-26:
`dhcp-option=option:ntp-server,192.168.28.7`

That would be it.
I would check the administration page in the routers web interface to make sure it is off there. Worst case could make a dnsmasq_postconf script to delete it.

RMerl, 2019-05-26:
Note that the DHCP option is different from actual interceptions.  The DHCP option is always present whenever the NTPD is enabled, and it's up to clients to decide if they want to use it or not.

Interception works at the iptables level, redirecting any outbound connection to port 123 to use the router's IP address instead.

This is not a bug, and working as intended.

closed 2019-05-26 by RMerl