RMerl/asuswrt-merlin.ng #315: OpenVPN CPU question

XbNz, 2019-06-05:
[image]

I have a 50/50 connection, when I use OpenVPN, that changes from 50/50 -> 20/50

I traced the problem to the CPU. For some reason, when it downloads, it only utilizes half of 1 core (I know OpenVPN is an inherently single-cored process). For upload, it uses all the CPU - which btw results in insane buffer bloat, but that's beside the point. My question is why does download use only half of 1 cores' resources while upload uses the whole thing, clearly resulting in higher speeds. 

If there is any script that I can run to force it to use all the resources it can, I'd appreciate if anyone lets me know. 

Router model: RT-AC1750_B1 (RT-AC68U for Merlin firmware - latest update)

RMerl, 2019-06-05:
The issue tracker is only for development purposes.  If you need technical support or have a question, please use the support forums at https://www.snbforums.com/forums/asuswrt-official.51/ instead.

closed 2019-06-05 by RMerl

XbNz, 2019-06-06:
Thanks, will do.