RMerl/asuswrt-merlin.ng #321: dhcp manual assignment changes require a reboot

bughit, 2019-06-16:
RT-AC68U_384.11_2

In the stock firmware, manual assignment change apply takes longer but after, the new config is active (dhcp renew picks it up)

In RT-AC68U_384.11_2 applying changes is quicker but they are not active after (dhcp renew does not pick it up), until a reboot.

RMerl, 2019-06-17:
Works correctly for me.

bughit, 2019-06-18:
- start in a state where MAC1 is mapped to IP1 and a machine with MAC1 has IP1
- edit the MAC1=>IP1 mapping to MAC2=>IP1, apply
- I believe with stock firmware this assigned a new unreserved IP to the machine, presumably because the router LAN interface was cycled
- with yours, machine retains IP1, presumably because the router LAN interface is not cycled
- change the machine MAC1 to MAC2
- with stock firmware the machine would get IP1
- with yours it gets a new unreserved ip, despite the MAC2=>IP1 mapping

closed 2020-09-21 by RMerl