RMerl/asuswrt-merlin.ng #326: WebUI loading issue - 384.12 on RT-AC87U

GeorgeWells, 2019-06-23:
My laptop on WiFi and also other WiFi connected devices are unable to load the WebUI on 384.12 properly after logging in, even across multiple browsers and the use of incognito/private windows. The login page loads instantly but when you try to log in it seems to hang for a long time (console tab in Chrome developer tools shows some JS files failed to load with ERR_CONNECTION_RESET) . The issue is less apparent on my main PC which is connected directly with a cable however it happens occasionally. Rolling back to 384.11_2 resolves the issue.

If there's anything I can do to help diagnose/resolve the issue please let me know, I am happy to help.

ademarre, 2019-06-26:
I am experiencing the same issue with 384.12 on RT-AC88U.

Chrome shows `net::ERR_CONNECTION_RESET`. It is usually a mix of .css and .js requests that fail to load. These seem to fail more often than others: `form_style.css`, `validator.js`, `state.js`. But sometimes the failure is on the main page request, such as `index.asp`.

No matter which requests fail, the network profile is similar. The download appears to begin with a reasonable TTFB, but then only a few hundred bytes are downloaded. Chrome eventually reports the failure about a minute later, and more often then not the page is not in a working state.

RMerl, 2019-06-26:
This has been the case for years when using https, and so far nobody has ever been able to track down the reason.  Since the issue mostly disappear once you use a certificate that's signed by a CA recognized by the browser, I suspect it's caused by the browser trying to do some kind of certificate validation in parallel.  Since the router's httpd is single-threaded, it causes deadlocks to randomly occur.

The only workarounds that are currently known are to either generate your own CA and sign your certificate with it, or to switch to http.

GeorgeWells, 2019-06-26:
Well I've learned something today!

I currently have the router set up to use http and not https (hopefully looking at the right setting):

[image]

RMerl, 2019-06-26:
Yeah, non-https access should be fine, as long you trust your LAN.

Someone more familiar with either TLS validation by browsers or socket code would have to look into this, as this is outside my field of expertise.

ademarre, 2019-06-27:
I access it with HTTP and have it configured for HTTP only. I am not using HTTPS. After installing 384.12, the web UI is slow and unreliable using HTTP.

master3395, 2019-06-29:
If you restart the router, the UI comes back.
Not sure what reset's after a stop of power, but my UI didn't even load.
After 2 weeks running, without restart.

GeorgeWells, 2019-06-29:
I have just tried this, unfortunately it didn't solve it for me.

cmaohuang, 2019-07-05:
@GeorgeWells I have the same issue with 384.12 on RT-AC86U.
But how do you roll back to 384.11_2 without WebUI?
or how do I set up to use http without WebUI? I appreciate that you could give me some idea.

j4yn, 2019-07-10:
I got the problem after I activated traditional QoS. After many restarts and long waiting for log-in, I managed to turn it off again and everything was fine. Http login was always activated.

kirk911, 2019-07-10:
I experienced this issue. Could reach login, but then it would timeout attempting to load the next page. I disabled https via SSH, but the problem persisted. Note that I have had https enabled for multiple versions prior with no issue. Rebooting would temporarily fix it. Reverted to 384.11_2 and everything is back to normal.

zyxtomatic, 2019-07-10:
I can confirm the same webUI issue on my RT-AC88U.  After upgrading to 384.12, the webUI would not reliably load beyond the login screen.  Sometimes, refreshing the page a couple of times would get each page to come up. Doing that page refresh dance, I was thankfully able to get into the firmware update menu in the gui.   Downgrading the firmware back to 384.11_2 solved the issue.

Additionally, I am not using HTTPS for the webUI due to the long-standing performance issues described above by others.

crkinard, 2019-07-13:
Any time I have had issues loading the web interface one of two things has fixed it. But I will add it is VERY rare for me to see this issue. Usually only when I am trying to access it from a computer that recently changed IP's.

1) Clearing the browser cache.
Or if its really buggered
2) SSH in and 'service restart_httpd'

j4yn, 2019-07-19:
I can reproduce it on my AC68U. Turn on traditional QoS -> cannot log in anymore after first logout. I have only Http on. No browser works (phone, tablet, other PC), also after clearing cache.

Eldenroot, 2019-07-20:
I can reproduce - same as @j4yn - when you enable QoS the webUI is slow and sometimes irresponsible.

Eldenroot, 2019-07-23:
Guys, here is the summary why this happens:

> I can reproduce this issue with QoS enabled. According to Merlin's issue discussion, someone mentioned that Merlin 384.11_2 has no such problem.
> 
> It's very likely caused by commit [f67ebf5](https://github.com/gzenux/asuswrt-rtn18u/commit/f67ebf59ef1314e56a237dec8c4c762980c3e238) and [dbd4abf](https://github.com/gzenux/asuswrt-rtn18u/commit/dbd4abf8f7773a24a14d3c04d63fac7b7a89e543), which solved CVE-2019-11477, CVE-2019-11478 and CVE-2019-11479.
> I can't figure out why these commits conflict with QoS feature and I'm not sure whether should revert these commits since they are security patches.
> 
> I would like to wait for Merlin to solve this issue. If you really need to enable QoS, my current suggestions are:
> 
> 1. Use previous release [384.10](https://gzenux.github.io/asuswrt-rtn18u/RT-N18U/RT-N18U_384.10_0.trx) (384.11 has already merged Merlin 384.12)
> 
> or
> 
> 1. Apply the following workaround to disable TCP SACK (needs to apply setting every boot)
>    `# echo 0 > /proc/sys/net/ipv4/tcp_sack`
> 
> Please note that the GUI response is slower than `#1` but a little faster than current state.(tested on my router)

link: https://github.com/gzenux/asuswrt-rtn18u/issues/8

RMerl, 2019-07-24:
I just tested it with my own RT-AC88U and Adaptive QoS enabled, and my webui is just as snappy as before.

Eldenroot, 2019-07-24:
It is quite fast when you enabled it but wait few hours and check again.

Everything is working and is fast excluding GUI.

themiron, 2019-07-25:
should be fixed with fe66c7b88f745ddb657ba8fa4abd48fe60880751, please check

RMerl, 2019-07-29:
@Eldenroot can you confirm if the issue is resolved with 384.13 beta 1?  The fix has been merged in it.

Eldenroot, 2019-07-29:
I am waiting for RT-N18u beta build from @gzenux - https://gzenux.github.io/asuswrt-rtn18u/

Unfortunately I cannot use yours official builds because of compatibility. I will let you know when I get a new build.

ademarre, 2019-07-30:
384.13 beta 1 appears to resolve the WebUI loading issue I experienced with 384.12 on RT-AC88U.

It's early to say for sure, but it may also have resolved an issue I've been trying to debug where I frequently see high latency on new connections. I first noticed it after reverting to 384.11_2 because of this WebUI issue. I didn't notice the problem with 384.11_2 the first time around.

barrystaes, 2019-08-02:
I had no webui https problem (using 384.12 on AC3200) but i did notice that WAN access was not working while enabled. I had to change the setting back and forth to make it work again.

closed 2019-08-20 by RMerl

joaomoreno, 2021-04-29:
@RMerl Thanks for all your hard work on this project, really appreciate it!

I just installed 386.2_2 on my RT-AC68U and I have literally the same symptoms:

- Web UI doesn't load at all after login
- `service restart_httpd` allows me to see the login screen once again, but I can't get past that
- Harware reboot will sometimes let me have a longer session. I was lucky to get one long enough to enable SSH and experiment a bit, but I haven't found anything that actually fixes the issue for good.

```
# nvram show | grep http
last_httpd_handle_request_fromapp=
last_httpd_handle_request=
https_crt_gen=0
https_crt_cn=
http_username=admin
http_lanport=80
http_passwd=XXXXXXXXXXXXXXX
le_acme_auth=http
apps_ipkg_server=http://nw-dlcdnet.asus.com/asusware/arm/stable
misc_http_x=0
http_clientlist=
webdav_http_port=8082
https_crt_save=1
misc_httpsport_x=8443
size: 56683 bytes (8853 left)
http_client=0
pushnotify_httplogin=1
http_autologout=30
http_id=TIDe855a6487043d70a
http_enable=0
https_lanport=8443
webdav_https_port=443
httpd_die_reboot=
misc_httpport_x=8080
http_dut_redir=0
```

Also, QoS seems disabled?
 
```
# nvram show | grep qos
qos_reset=0
qos_irates=100,100,100,100,100,0,0,0,0,0
qos_rst=off
qos_sticky=1
qos_rulelist=<Web Surf>>80>tcp>0~512>0<HTTPS>>443>tcp>0~512>0<File Transfer>>80>tcp>512~>3<File Transfer>>443>tcp>512~>3
qos_overhead=0
qos_orules=
qos_ibw=
qos_type=1
qos_obw1=
qos_syn=on
qos_burst0=
qos_ack=on
qos_burst1=
qos_method=0
qos_atm=0
qos_icmp=on
size: 56683 bytes (8853 left)
qos_enable=0
qos_obw=
qos_default=3
qos_orates=80-100,10-100,5-100,3-100,2-95,0-0,0-0,0-0,0-0,0-0
qos_ibw1=
qos_fin=off
qos_bw_rulelist=
qos_rulenum_x=0
qos_mpu=0
```

joaomoreno, 2021-04-29:
Nevermind, this seemed to have been a temporary issue. 4 hours later and this started to work. 🤷

TakSawa, 2021-06-05:
Well. I got an RT-AC68U. Flashed 386.2_4 today. And got the same problem.
At first, I was suspecting that factory resetting my router will solve the problem. And it did.
I changed wireless parameters and hid the ssids. I did nothing else. I come back one hour later and the gui won't load again. :-(
I flashed Merlin to try to resolve an issue with the scheduler. If I can't access the gui, what's the point ?
Sorry for the rambling. But it's been a long week. Router problems are the last thing I wanted to experience tonight.

r14chat, 2021-06-09:
> Ever figure out what caused it?
> Just flashed my AC3100 to 386.2_6 today and now I can't access the panel at all.
> Sucks because SSH is disabled, and I don't want to reset as I have countless firewall settings.
> _The one time I don't backup settings..._

I believe it to be related to this:
`386.2 (2-Apr-2021)
  - NOTE: due to changes in how custom device icons are handled,
          first time you boot with 386.2 you need to either
          shift-reload the main index page, or clear your
          browser cache.`

The first load took quite awhile, regardless of cache/browser/device. After letting it go (weird, no timeout for an hour) the first time, then shift reloading, there are no issues.

we1559, 2021-07-03:
I just upgrade my ac68u to 386.2_6 and I meet the same problem.
I can login in the web UI the first time after the router restart. But the page just hang and chrome shows it didn't response.
I can't login in the web UI by the chrome, edge, IE in my computer which connect by lan. 
I'm also not able to login in the web UI by my phone.

It seems to be a big problem.